balena-io/etcher #1426: Etcher is corrupting SD and USB drives.

8BitSloth, 2017-05-14:
- **Etcher version: 1.0.0**
- **Operating system and architecture: Arch Linux X86_64**
- **Image flashed: Raspbian Lite**
- **Do you see any meaningful error information in the DevTools? No**

Etcher has corrupted two brand new SD cards and one USB drive. I formatted a new 64 gig SD drive to fat 32 using gparted then flashed Raspbian to the drive. I decided to use Raspbian Lite and wrote it without a reformat. The flash failed and now the drive cannot be modified, though the files can still be read. For giggles, I tried a second drive that was a new 8 gig drive. Again I formatted, flashed, it worked, reformatted, flashed again, it worked, did not format, flashed and it's corrupted like the first. I decided to try a cheap Lexar USB drive and the exact same thing happened if I try to flash twice without reformatting first.

I've tried the Linux and Windows fix suggested in the docs and it doesn't work. I tried flashing a different image, it doesn't work. I tried formatting on another PC and in windows, it doesn't work. None of my drives were bad before this, two were fresh out of the box, and one was a 5 month old USB drive. For me the issue happens like clockwork with SanDisk and Lexmark drives. I would love to provide logs, but after being disgusted with myself for choosing this program over Imagewriter that I shutdown and about threw my laptop into the road.

lurch, 2017-05-15:
I'm so sorry to hear that you've had a bad experience with Etcher so far. I know it probably doesn't help you much, but I can assure you that for most users it works absolutely fine.

For the SD cards, it's always worth checking that the write-protect switch hasn't accidentally moved into the 'locked' position. What errors do you get when you try running the `dd` instructions from https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#gnulinux ? Are you able to 'repair' your drives using the official [SD Formatter](https://members.sdcard.org/downloads/formatter_4/) tool on Windows?
Are you using an internal or an external SD card reader? (sometimes USB card readers are more reliable than internal readers)

In "normal operation" there shouldn't be any need to reformat the cards / USB drives before writing an image to them with Etcher, it should "just work" regardless of what was on the cards previously.

Are you sure your cards are "genuine", or is it possible that they're fakes that you've bought e.g. from eBay? If you want to be certain, you could try testing them with F3 (on Linux) or H2testw (on Windows).

MrTeofil, 2017-05-21:
Hello
I have the same problem
It says that my card is 128 Mb instead of 14,6 Gb. I tried [SD Formatter](https://members.sdcard.org/downloads/formatter_4/) but it didn't work.

By the way, the SD card is new, i wrote on it an image with [Win32 Disk Imager](https://sourceforge.net/projects/win32diskimager/) and it worked, but I saw this program on [Raspberry Pi Installation Guide](https://www.raspberrypi.org/documentation/installation/installing-images/README.md) page and I decided to try it.
Is there any solution to fix this SD card?

jviotti, 2017-05-21:
Hi @MrTeofil,

> It says that my card is 128 Mb instead of 14,6 Gb

That's normal. Images include their own partition table, so if the image has a partition of 128 MB, then the OS will detect its size as 128 MB.

> I tried SD Formatter but it didn't work.

Did you try the Windows formatting instructions from the documentation? https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#windows

lurch, 2017-05-22:
@MrTeofil 
If Windows is reporting the card size as 128 MB, then I suspect @jviotti 's diagnosis above is correct.

But if it's Etcher itself that's reporting the card size as 128 MB, then you may be suffering from this bug https://github.com/resin-io-modules/drivelist/issues/142

(The reason for the different 'behaviour' between Windows and Etcher, is that Windows only reports the size of the first partition on a drive, whereas Etcher reports the whole size of the raw drive, regardless of what partitions are on it. This is because Windows writes to a drive at the filesystem-level, whereas Etcher writes to a drive at the lowest whole-disk / raw level)

MrTeofil, 2017-05-28:
@jviotti @lurch 
Thank you, it worked !

> Did you try the Windows formatting instructions from the documentation? https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#windows

Sorry if I bothered you.

jviotti, 2017-05-29:
No worries, glad it worked! :)

closed 2017-05-29 by jviotti

jjh221, 2018-04-05:
Etcher did the same thing to me today. Could not format. I put the SD card in my camera to format then could format with windows. Cheers

Patchers, 2018-04-08:
Did it to me today. :( not happy about etcher

Patchers, 2018-04-08:
I put the SD card into diskpart and my phone in order to fix it and nothing has worked

maltebeckmann, 2018-04-23:
After flashing my microsd card with Etcher, it doesn't show up anymore and get very very hot. Although I just find it very hard to believe that software could break hardware.

jhermsmeier, 2018-04-23:
@Patchers @maltebeckmann can you tell us what OS you're on, which Etcher version you used and what image you flashed? We can't tell you much without any info :)

maltebeckmann, 2018-04-23:
Thanks for getting back to me! Running Etcher Version 1.3.1 on OS 10.13.4. I flashed 2018-03-13-raspbian-stretch-lite onto the microSD. I did it 2-3 times over the course of an afternoon because the boot partition wasn't showing up that would allow me to set up the Raspi without an external screen. Now the card doesn't show up anymore and gets incredibly hot. I am going to buy another one tomorrow to test this.

adiluxx, 2018-05-03:
hi, I'm having a similar issue, I've tried flashing Chromium OS on 2 different thumb drives (1 brand new, 1 used) both have same results when I try to clean using diskpart.  I had to reflash these drives because i figured i had the wrong architecture because these flashed drives couldn't boot, but was recognized by BIOS.
DISKPART> list disk

  Disk ###  Status         Size     Free     Dyn  Gpt
  --------  -------------  -------  -------  ---  ---
  Disk 0    Online         1863 GB      0 B        *
  Disk 1    Online          558 GB      0 B
  Disk 2    Online           14 GB    14 GB

DISKPART> select disk 2

Disk 2 is now the selected disk.

DISKPART> clean

DiskPart has encountered an error: Incorrect function.
See the System Event Log for more information.

DISKPART>

cj2tech, 2018-06-12:
I know it’s closed but I’ve had problem with windows 10 corrupting SD cards. I fix them with at formatted and doing a full format instead of a quick on.

Ewpratten, 2018-07-02:
I have the same problem. etcher has messed up 12 of my drives and 2 sd cards

jhermsmeier, 2018-07-02:
@adiluxx the "DiskPart has encountered an error: Incorrect function." hints at an issue with Windows; you can try the `rescan` command in diskpart to refresh the state of attached storage before running other commands – if that doesn't help, there might be something wrong with the OS (i.e. corrupted components, failed chkdsk)

franknfurther, 2018-07-06:
I have today destroyed two sd cards using Etcher, and I used in on two different computers so I know its not caused by the hardware.

jhermsmeier, 2018-07-06:
@franknfurther Etcher doesn't destroy any hardware, it "just" copies the disk image to whatever device you choose – if your operating system doesn't recognize any file systems on the flashed device (or fails to format it), it's still not broken. While we've occasionally seen cases where the hardware does give up; it was due to counterfeit, faulty, old hardware or malfunctioning SD-card readers / USB hubs etc.

> this needs some more serious testing before being released to the world

Believe it or not, we do actually test what we make, and use it every day on countless devices – and so do a couple hundred thousand other people on this planet. Doesn't mean it's bug-free though.

lurch, 2018-07-06:
https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

franknfurther, 2018-07-07:
I hear you, but how do you explain that I destroyed two perfectly working SD cards, using two different Windows 10 PCs, by using Etcher? Just because I could not believe it myself, I went to the store to buy another card and used the old win32 disk imager on one of the above mentioned PCs. And it worked. Is it coincidence?

lurch, 2018-07-07:
AFAIK Etcher and Win32DiskImager essentially do more or less the same thing (they just use different technologies internally, and Etcher obviously has a much nicer UI) - so I suspect any cards that got "broken" by Etcher would also get "broken" by Win32DiskImager, and any cards that "work" with Win32DiskImager would also "work" with Etcher.
Although I'm sure @jhermsmeier would be very interested to hear if you had a reproducible test case...

franknfurther, 2018-07-08:
I agree, two corrupted sd cards in one day do not make a large enough sample to blame Etcher.  I am out of SD cards now but will definitely use Etcher again next time and let you know if it doesn't work. Hardware is about 1-3 years old, we are talking Intel and Lenovo PCs so that should be ok. But I might need some advise regarding which SD card brands are ok and which ones are not.

adiluxx, 2018-07-08:
4 corrupted USB drives.  Some were v easy to fix, but i've tried disk part so many times, and system tells me that the drive is busy and can't clean the drive.  I think i've finally managed to recover the last of my flash drives by using chkdsk/Format, and other methods from windows 10, 7, machines and from using a partition software, but i'm not using Etcher anymore, gave me more problems than good results.  There are many other software that do the job without corrupting flash drives.

https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

-adilux

lurch, 2018-07-09:
> system tells me that the drive is busy and can't clean the drive.

See also #1793 and #1991

rmbrown410, 2018-07-20:
I just deleted the VM with the log file on it so I can't share them but it just destroyed one of my thumb drives. I was trying to create a bootable USB from an Ubuntu Server 18.04 ISO on a VM running macOS Sierra 10.13.5. (Etcher version 1.4.4) I totally gave Etcher a shoutout via Facebook last time I used it because it ran right, it was compatible with macOS and capable of imaging ISOs in a convenient fashion. This experience.....not so good. If the application relies on the dd command, that might be the source of the issue. Although I'm not sure of another method to image an ISO to a drive.

lurch, 2018-07-24:
I don't know what Etcher has to do with Android smartphones? :confused:

timothytoddbrandt, 2018-08-01:
Had same issue.  The only way I could get the USB drive to format full size was to begin a format, not a quick format, and physically pull the USB out halfway through formatting.  That forced Windows to recognize it as unformatted but full capacity.

bendibor, 2018-09-16:
Same thing happened to me today. Never had an issue with other similar software. No matter what I do, the drive stays broken. When I read all this I get the feeling Etcher doesn't want to hear it, but this needs fixing!

jviotti, 2018-09-17:
@timothytoddbrandt @bendibor This is a very common misconception and Windows issue, and we had explained this many times before.

Etcher is **not** corrupting drives. When you flash an image which comes with a file system doesn't recognise (i.e. almost every GNU/Linux image out there), Windows won't be able to detect it because it doesn't recognise that file system (as expected). Other flashing programs that specialise on GNU/Linux images, like UNetBootin, don't flash the image byte by byte, like Etcher, but re-format the drive and move files around so support very Linux specific features, sometimes using a format Etcher recognises. These are two types of different type of flashing, and Etcher specialises in the former.

Now Windows' formatting is tool is known to be very bad, and to fail at formatting drives without a known file system with strange errors. I suggest using the `diskpart` tool directly (works almost every time).

In summary, Etcher is not corrupting any drive; its just the OS that fails to recognise the new file-system in the drive. If this is a GNU/Linux image, please try to boot it, or read it from a GNU/Linux system, and you will see that it works fine.

jviotti, 2018-09-17:
Please see https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives for more details.

jviotti, 2018-09-17:
> The only way I could get the USB drive to format full size was to begin a format, not a quick format, and physically pull the USB out halfway through formatting. That forced Windows to recognize it as unformatted but full capacity.

OS images come with a partition table of a fixed size. If the image says it contains a 1 GB partition, when you flash it, then the resulting drive will contain a 1 GB partition, no matter if the drive is 2 GB, 4 GB, or more. That's why if you re-format the drive, it will clear the old fixed partition, and you will see that it has "full size" again.

jviotti, 2018-09-17:
@rmbrown410 

> This experience.....not so good. If the application relies on the dd command, that might be the source of the issue. Although I'm not sure of another method to image an ISO to a drive.

Etcher **does not rely on dd** or on any other external tool. It implements a byte-by-byte flashing algorithm in JavaScript. 

As mentioned before, there are two ways image flashing programs work:

- *Re-formatting and copying*: Like UNetBootin. They will re-format your drive, create their own partitions, and copy files inside the original image to the drive. This means that they can re-use the full size of the drive, implement persistence mode, and make it like the program didn't "break" your drive. On the other side, these techniques are OS specific (i.e. they won't be able to flash BSD images or any other OS) **and they modify the source image, so you don't flash exactly what upstream OS developers created**
- *Byte-by-byte flashing*: Like Etcher and `dd`. This flashes **exactly** what the image contains, byte-by-byte, which gives you a lot more safety and is independent of the OS you actually flash

Both types of flashing have their own benefits, depending on what you're after. Do keep in mind that if you chose the latter, your OS might not be able to recognise the resulting drive, which is completely normal and expected.

jviotti, 2018-09-17:
@jjh221 

> Etcher did the same thing to me today. Could not format. I put the SD card in my camera to format then could format with windows. Cheers

Don't you think its weird that your camera was able to format the drive but Windows wasn't? Don't you think this points out an issue on Windows rather than on Etcher?

jviotti, 2018-09-17:
I honestly hope these series of comments clarify what Etcher is and is not doing. It is really frustrating as a contributor to see how Etcher gets thrown a lot of shit in different mediums because of technical misconceptions, over and over again, even though we took a lot of effort to clarify this to hundreds of users, created entries in our documentation, and even explained this whole thing in this thread that still keeps receiving a lot of bad comments.

I understand that not everyone is super technical, and you are definitely not expected to know this from the start. But PLEASE, take a moment to read what we wrote, try to understand, and ask peaceful questions rather that writing "not happy about etcher", "etcher has messed up 12 of my drives and 2 sd cards", "this needs some more serious testing before being released to the world", "i'm not using Etcher anymore, gave me more problems than good results", "I get the feeling Etcher doesn't want to hear it", etc.

We're building this software for free, as open source, for our users, and we don't deserve this treatment. 

There might be exceptional cases where a drive really ends up fried and we would love to get to the bottom of it. While it might not be something that we can "fix" in Etcher (Etcher uses the same Windows OS APIs as every other Windows app, so we don't really have any special control over things), we can build protection inside Etcher so that we improve our error messages, or implement some sort of diagnosis step that can detect faulty drives.

Thanks a lot!

bendibor, 2018-09-17:
@jviotti I think you all need to be applauded for creating the software in the first place, so I get why it's frustrating to read the complaints. But to me as a user, it seems an obvious issue that a 'Windows' program creating usb drives 'safely and easily', in this process makes the drive inaccessable and for many users inpossible to use/recover on Windows later on. It is Etcher that without a clear warning, does this. Yes ofcourse, flashing has advantages... but this is a disadvantage user should know about. 
When so many users run into trouble after trying Etcher, instead of blaming them and 'stupid windows', isn't it a good idea to either provide future users with a warning or provide a 'recover usb for windows mode'? Because for me and many others, the provided help pages don't work. Only after spending a day reading and trying all kinds of different solutions, in command line and software I was able to find a solution that did. Not fun. It's not your fault, you are great but at the same time you could have prevented ignorant users like me from landing in this ordeal ;)

lurch, 2018-09-24:
See also #273

jviotti, 2018-09-24:
@bendibor 

Thanks for the kind words!

> But to me as a user, it seems an obvious issue that a 'Windows' program creating usb drives 'safely and easily', in this process makes the drive inaccessable and for many users inpossible to [...]

This is not a "disadvantage" of flashing, its just the way it is. There are some ways to do it differently in some cases, but the definition of flashing an OS image is to write an OS to a device, and chances are that the OS will use file systems your host OS doesn't recognise.

> When so many users run into trouble after trying Etcher, instead of blaming them and 'stupid windows', isn't it a good idea to either provide future users with a warning or provide a 'recover usb for windows mode'? Because for me and many others, the provided help pages don't work.

That's a good point. We do something like this to show a warning dialog when people try to flash Windows images with Etcher, as these images are not like other images, and thus a traditional flashing won't work, so we can inspect the partition table, and potentially show a warning at least when flashing Linux images on macOS and Windows cc @thundron.

> Only after spending a day reading and trying all kinds of different solutions, in command line and software I was able to find a solution that did. Not fun

The word "solution" still implies there was a problem to fix. Your end result is the expected result. I do agree that some OSes (specially Windows) makes it super difficult to revert back to another file system, which is a great point. As @lurch pointed out, something like https://github.com/resin-io/etcher/issues/273 would help a lot. I believe this is a very easy feature, so maybe worth putting some time on it @thundron @zvin ?

thundron, 2018-09-25:
@jviotti I will let you know, but on paper we can surely add a button to wipe a selectable drive, it could also be something along the lines of https://github.com/resin-io/etcher/issues/1426#issuecomment-421962780 , @zvin ?

bendibor, 2018-09-25:
@jviotti
I guess we only have to agree to disagree on the use of the words disadvantage and solution. 
In my line of work if a user comes to me with a 'problem' it wouldn't cross my mind to say it's really 'not' a problem, as apparently it's a problem to them and I try to take my users serious even though opinions may differ. 
Also saying something is the way it is seems to always imply there isn't a possible way to change it. Well, there is, even in the unlikeliest circumstances:
https://s.s-bol.com/imgbase0/imagebase3/large/FC/6/7/1/7/9200000077297176.jpg

jviotti, 2018-10-01:
@bendibor I understand that, and it's the reason why I suggested the warning above. The problem in this case is that the issue happens outside the scope of Etcher (i.e. the OS file-system and mounting mechanisms) on which we don't have any control over, which means is almost impossible for us to do anything else other than displaying a warning. We could modify the image's partition table to provide a readable first partition, but we would be modifying upstream images, which is what Etcher is trying to prevent altogether.

bendibor, 2018-10-01:
@jviotti The warning is great, all it needs! And perhaps a referral to the help page on how to 'restore' (even if not broken ;) a drive so windows can see it again. The suggested wipe button should be able to do that trick perhaps?

lurch, 2018-10-03:
I guess if a "wipe" button is added to Etcher (i.e. overwriting the MBR and/or GPT with zeros), you need to make it clear that this is different to "format", and the user will still need to use their OS's formatting tools before they're able to use it as a "normal" USB drive again.
(as on Windows, the drive may need to be formatted as FAT16, FAT32, NTFS or ExFAT, depending on the drive's size and physical medium. And of course OSX and Linux have their own native FSes too)

bendibor, 2018-10-03:
Good one

Ewpratten, 2018-10-03:
Overwriting the drive with zeros has always fixed my problems. Having a
button to do it would be good.

kureci, 2018-10-21:
This software should really be stopped! It's corrupted multiple of my SD cards, it's proving impossible to get the back to normal.

jviotti, 2018-10-22:
@kureci Have you read all the comments in this thread? Etcher is literally un mounting the drive, and writing bytes to it using the typical OS system calls. There is no way we can corrupt SD Cards or "write protect" the drive so you can't write to it. It doesn't make any sense to request an open source software to stop because you're having troubles with it. You are more than welcome to read the source code if you don't trust it. 

Check if the partition table from the image that you are flashing declares certain partitions as read-only, or the SD Cards that you are flashing to. Have you tried zeroing the drive with `dd`? Try:

```sh
dd if=/dev/zero of=/dev/rdiskN bs=4M
```

If that fails, please paste the error information.

kureci, 2018-10-22:
@jviotti the fact that so many people complain about it corrupting their SD cards means it's more than just me "having troubles with it". Of course no software is perfect and it might be edge cases that Etcher is causing these issues.
You say Etcher is only calling OS commands to write the bytes, yet when I've used `dd` on Mac in the past I haven't come across this issue of write protection.

Anyway, thanks for the suggestion. I tried running `dd if=/dev/zero of=/dev/diskN bs=4m` but getting:
`dd: /dev/disk2: Permission denied`

It's not a user permission issue as I've tried running with and without `sudo` - I've checked the lock switch on SD Card adapter, no luck.

jviotti, 2018-10-22:
@kureci We have only a handful of people complaining about this problem compared to amount of successful flashes Etcher reports through opt-out analytics (>100000 flashes every month). I agree there is always the possibility of an actual bug, which would be affecting a really really small percentage of our users. 

We had many reports like his (mainly on chat), and after digging a bit, turns out to be a confusion about file-systems in different OSes, faulty SD Cards, or some problem with the re-formatting tool they were using, for whatever reason. Keep in mind that one of the main roles of an operation system is to protect client code from causing damage to the hardware. The OS presents an interface (i.e. `umount()` and `write()`) that both `dd` and Etcher uses that is designed to prevent software like Etcher from doing whatever it wants and breaking devices. We don't have access to external devices without going through the kernel first, so the chances that we are really breaking devices are very very slim (i.e. an OS or driver bug in one of the system calls we make).

Regarding `dd`, did you try un mounting the disk before running `dd`? If you are on macOS, you can do `diskutil unmountDisk /dev/disk2`

kureci, 2018-10-24:
@jviotti tried unmounting first, erasing using `diskutil eraseDisk`. I've also tried using a USB adapter instead of a SD Card adapter and even though `dd if=/dev/zero of=/dev/diskN` zero-ed out the whole SD card by writing 30GB worth of zeros in there, I open the SD card and the previous files are still there! Nothing has changed. Next step is to order an SD card and hammer these to dust!

I just remembered, I've used these SD cards on Asus tinkerboard, I wonder if it's that that's corrupted the SD cards! Will look there too

lurch, 2018-10-24:
> even though dd if=/dev/zero of=/dev/diskN zero-ed out the whole SD card by writing 30GB worth of zeros in there, I open the SD card and the previous files are still there!

That sounds very very strange. Is it possible the SD cards themselves are faulty? Maybe it's worth buying some SD cards from a different manufacturer, to rule out the problem being with Etcher (or your computer) rather than the SD cards themselves?

ShiftLeftLogical, 2018-10-24:
Ok. I have just caused two different cards to go from fresh out of the package working to seeing on linux/windows 'no medium' / 'no media' for both mount / disk management tools, respectively. This is Etcher on windows.

I figured the first card was just bad so I moved to a different card / manufacturer, same results. Flashing appeared to succeed and so did verification of data but then the card reader / usb media drive is no longer recognized as even having media installed.

Is there any logging mechanism I can submit to help understand what happened and why I cannot see these media devices?

Note that I have used Etcher just fine with this personal card reader/writer on my personal windows machine w/o a single issue, this is not my windows machine so I am not sure if there might be configuration details of the OS which are impacting this (patches/reg/etc.) Are there any registry requirements which are known problems? Are there any BitLocker setting which are known problems?

kairsas, 2018-10-25:
> @kureci We have only a handful of people complaining about this problem compared to amount of successful flashes Etcher reports through opt-out analytics (>100000 flashes every month). I agree there [...]

Same problems for me. 
After first use of Etcher i am fighting *Permission denied*, no success with reformatting or dd or any other ways.
And it's not OS user issue.
Never experienced that kind of problems before.

Scripting-the-Void, 2018-10-26:
Just to add in my own recent story to this thread: I've also run into issues recently using Etcher for loading Proxmox onto a USB. It corrupted my drive and Windows formatting tools weren't able to do anything with it after a flash had failed. Since Windows wouldn't even poke at it from a distance, I opened Rufus and tried running a flash with some other ISO file I had laying around to see if Rufus could work its magic. It worked and the drive is now usable again. Looking forward to using Etcher because Rufus won't flash the ISO file I want.

99sono, 2018-11-24:
Following the tutorial:

https://tutorials.ubuntu.com/tutorial/tutorial-create-a-usb-stick-on-macos?_ga=2.110392581.1832303617.1543043795-1884538113.1543043795#2

It looks like i just managed to get a Fried up USB 8 GB pen.

The make disk utility to rescuse the device seems not to work at all.

I would recommend everyone to not use this tool.
One thing is the flashing process to fail and need to be repeated.
Another one is for a storage device to become irrecoverable.

amquei, 2018-12-07:
I think thal I solved this problem using GPARTED and creating a new partition table (msdos) on the my pendrive several times until worked.

lurch, 2018-12-27:
Interesting to see that authors of other disk-writing software also get just as many unsubstantiated complaints about the software "breaking" their drives https://github.com/pbatard/rufus/issues/313
:roll_eyes:

kureci, 2018-12-27:
@lurch interesting choice of words there.. "unsubstantiated" complaints? I bet you think complainers here are conspiring to complain about all such software. Dude, if you're a contributor and not willing to see what could be causing this, at least don't just talk nonsense. What "substantial" evidence do you need to believe that this software has corrupted so many sd cards? Send me your address and I'll post my two 32gb U-class high quality microSD cards that have been corrupted by Etcher. If you manage to fix them then they're yours to keep.

thundron, 2018-12-28:
@kureci What do you mean by "corrupted"? What's the issue you're having with them? Do you have any logs? Did you try solutions provided for different OSes [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md) as well as reading through the whole document to verify you're not in one of those cases?

I hope you can now easily see why @lurch referred to these as "unsubstantiated complaints" and I'd advise you to read [this](https://github.com/raspberrypi/documentation/issues/1042) issue as well (the last 3 comments in particular) that might enlighten many aspects of users that got corrupted sd cards "by Etcher".

kureci, 2018-12-28:
@thundron if you go back in comments you can see that I've tried a few things, tried to put relevant bits of information regarding what the output is.. etc.

I won't be using Etcher again, and not because it has issues (even if minor), but because its developers seem to be denouncing any kind of claim that Etcher is causing SD card corruption. Everywhere I read comments from its promoters, the first thing they do is deny Etcher having anything to do with it, rather than listen and perhaps think of any possibility. Because clearly Etcher isn't just a UI on top of simple commands like `dd` on Unix, because I have used the `dd` command on the same machine numerous times in the past, with no significant issues.

Almost everyone who replies to these complains says "Oh Etcher is being used by thousands of people, if a few people are having issues with it then it's not Etcher" - seriously? OK, you can say it works 99% of the times, but at least be accepting enough to just consider the 1%

Anyway I'm not expecting to really have any useful output from this issue or the conversation. If I do get any spare time I suppose I could check the source and see if I can spot anything, otherwise I'll just not touch it again.

kureci, 2018-12-28:
Oh and, I've read a few comments from promoters/developers claiming that software cannot possibly "break" hardware - they probably think of the word break as some sort of smashing the cards into pieces or something.. goes to show how open-minded they are to accepting that software could be "damaging" the hardware not in a physical way, but in a way that makes other software impossible to write to it. Hardware isn't damaged as per se, but these SD cards aren't to be opened up and their circuits manipulated by hand to check that all parts are in place and that Etcher hasn't moved any parts!

[18 more replies]