balena-io/etcher #1427: Feature request: show flasing percents in the window headbar

TotalCaesar659, 2017-05-14:
Hello, could you add flashing percents to the window headbar? It's not comfortable to check progress manually.
[image]

lurch, 2017-05-15:
I guess you mean dynamically changing the title bar to e.g. `Etcher - Flashing 32%` or `Etcher - Validating 64%` ?
What do you think from a UX perspective @konmouz ?

jviotti, 2017-05-15:
Sounds good to me, but I wonder if this is a common practice for desktop apps. Let's get feedback from the design team.

konmouz, 2017-05-15:
@lurch & @jviotti This is a common practice in many applications. It is pretty much what we already do with the app icon and the progress bar (at least on mac). Sounds a nice addition to me.

jviotti, 2017-05-15:
OK, cool, let's go for it!

PR #1438 (2017-11-14): feat(GUI): add progress and status to window title
We add the progress percentage to the window alongside the status
(validating, flashing).

Closes: #1427
Fixes: #1439
Changelog-Entry: Add the progress and status to the window title.
closed 2017-11-14 by Shou