balena-io/etcher #1435: unsafe selection of target on Raspberry Pi

RogueM, 2017-05-15:
- **Etcher version: 1.0.0**
- **Raspbian Jessie on Pi3 (armv7)**
- **Image flashed: N/A**

on Raspberry Pi, Etcher 1.0.0 (compiled from source on Pi3 i.e armv7) is quite happy to write over the live filesystem in fact it is the default target:
[image]

... taking one for the team I went ahead:
[image]

... with a predictable outcome:
[image]

let me know if you need more details but I believe it would be a universal failure to safeguard the live system.

lurch, 2017-05-15:
Yeah, that's a suggestion I made before (see https://github.com/resin-io/etcher/issues/888#issuecomment-263355732 and https://github.com/resin-io-modules/drivelist/issues/126 ) which hasn't been implemented yet.
Closing this as a duplicate, and it's probably low-priority for now because Etcher doesn't officially support the Raspberry Pi yet.

lurch, 2017-06-08:
There's been talk recently about producing ARMv7 builds of Etcher, that would run on RPi2 and RPi3 boards. However IMHO we definitely ought to fix this bug (i.e. stop Etcher detecting the SD card that the Pi booted from as "removable") before we officially release an Etcher version that runs on the Raspberry Pi.

So re-opening this bug.

RogueM, 2017-06-08:
while not a fully comprehensive solution, can't etcher just deny user going ahead if the image to burn and the target partition is the same? I mean, it is not fully robust, but it should cover most typical case on the Pi?

RogueM, 2017-06-08:
in fact, on my Chromebook is already does in version 1.0.0, so I'm not sure it would be any different on the Pi?

jviotti, 2017-06-08:
> can't etcher just deny user going ahead if the image to burn and the target partition is the same?

This is already the case: https://github.com/resin-io/etcher/pull/1066

Could you give it another go? If the issue is reproducible, can you send us the following info?

- `lsblk -b -d "/dev/<device>" --output SIZE,RO,RM,MODEL`
- `udevadm info --query=property --export --export-prefix=UDEV_ --name="/dev/<device>"`

lurch, 2017-06-08:
@RogueM There's a difference between "source drive" and "operating system drive".
Consider a Raspberry Pi, booted from an SD card, with 2 USB sticks plugged in. The first USB stick contains the image you want to flash, and the second USB stick is the 'blank' one you're flashing the image to.
**Ideally** after selecting the image from the first USB stick, Etcher would then only offer the second USB stick as the available target drive.
_However_ while Etcher will identify the first USB stick as the "source drive", it won't identify the SD card that the Pi booted from as the "operating system drive" (that's what  https://github.com/resin-io-modules/drivelist/issues/126 is about), and therefore in the scenario described here Etcher would offer both the second USB stick **and** the internal SD card as possible destination drives.

Caveat: I haven't yet run Etcher on a Raspberry Pi myself...

RogueM, 2017-06-08:
Yes, as stated its's only one safeguard. My point was that the large majority of users will likely download the image in their home dir, and at the very minimum this should be picked up... 99.99% that was my case, and it didn't. But I'll try to confirm when I have a chance just in case the ISO was not in my root partition when I took the screenshot.

lurch, 2017-06-08:
Thanks @RogueM . In addition to the commands that @jviotti has requested above, if you can report the result of `mount` too that might be helpful.

closed 2018-10-15 by jviotti