balena-io/etcher #1439: Flashing / Validating inconsistent progress-bar display

lurch, 2017-05-16:
While looking at #1438 I realised that when we're flashing an image, the text on the progress-bar says just the %age e.g. `42%`, but when validating an image, we display e.g. `38% Validating...`.
The code for this is all in https://github.com/resin-io/etcher/blob/master/lib/gui/pages/main/controllers/flash.js#L103

Would it be more consistent if we displayed e.g. `42% Flashing...` on the progress-bar instead of just `42%` @konmouz ?  If this is all currently "as intended" though, then feel free to close this issue as "wontfix".

konmouz, 2017-05-17:
@lurch I agree with you

lurch, 2017-05-17:
Alternatively, instead of displaying `42% Flashing...` and `38% Validating...`, would it make sense to change it to `Flashing 42% ...` and `Validating 38% ...` to better match #1438 ?  \*shrug\*

Shou, 2017-05-25:
I was going to say I like how "42% Flashing" looks more, but the side effect is that "Flashing" will move about as the numbers change, whereas with "Flashing 42%" only the numbers 'move'.

PR #1438 (2017-11-14): feat(GUI): add progress and status to window title
We add the progress percentage to the window alongside the status
(validating, flashing).

Closes: #1427
Fixes: #1439
Changelog-Entry: Add the progress and status to the window title.
closed 2017-11-14 by Shou