balena-io/etcher #1440: removable drive not detected

mtantaya, 2017-05-16:
- Etcher-1.0.0-win32-x86
- windows 7 32bit ultimate sp1
- raspbian jessie
- *yes, at the first starting. here is the error i capture it
[image]

...........................................................................................................................................................................................
and... my removable device is not detected
[image]

thanks for your support and help

lurch, 2017-05-16:
Looks like a duplicate of #1054 ?

jviotti, 2017-05-16:
Indeed. The issue is that if the drive detection script fails, then it won't attempt to scan again (so we don't get hundreds of similar issues).

@mtantaya We're in the process of re-implementing the drive scanning routines in C++, which should hopefully fix this issue.

BTW, are you running any anti-virus software? If so, which?

closed 2017-05-16 by jviotti