balena-io/etcher #1442: Echet distroy my USB flash memory stick when install Alpine Linux !!

Nokia808, 2017-05-16:
- **Etcher version:**
1.0

- **Operating system and architecture:**
Linux Fedora X64 Cinnamon edition

- **Image flashed:**
Alpine Linux .iso extended edition version 3.5.2 X64 bit

Hi I tried version 1.0 of your program with 100% healthy USB flash memory stick of 32 GB capacity. I tried Fedora 25 without any problem. Then I format USB flash stick then tried to install Alpine Linux extended edition:
https://www.alpinelinux.org/downloads/
I downloaded iso image & check it's SHA256 & it was O.K Then installed it by Etcher. Then I tried to boot from this USB. Live USB boot run 1st O.K but stop asking me to enter credentials (user name & password) & this indicate error in flashing process. So, I tried to format my USB. But I'm unable know to format my USB flash stick !!!!!!!!!!! 

On Windows 7 I can not open USB flash to see it's contents at all after loading Windows ! I received message said "You need to format this device before you can open it". But when I tried to format it I received message said "Windows can not format this device. Check if it read only device".  

On Linux Fedora I can not open USB flash stick after loading Fedora by using file manager GUI !! It does not appear at all !! I just listen sound of plugin/plugout USB flash !! I can NOT format it by GNOME Disk: when I tried to format it by Disk, I received error message say: "This partition can not modified because it contain a partition table, please reinitialized layout of whole device. (udisks-error-quark, 11) see attached screenshot.

Please:

1) fix this fetal error: give support for Alpine Linux. It is the only Linux immune against rootkits attacks due to it use grsecurity in it's kernel ............

2) help me to restore my USB flash memory stick. It is now completely useless ! Please I'm new to linux (about 4 months to be 1 year user of Linux)

Best
[image]

lurch, 2017-05-17:
> help me to restore my USB flash memory stick. It is now completely useless !

Have you tried following the instructions at https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives ?

> this indicate error in flashing process

Did you leave validation enabled in the Etcher settings screen? If Etcher's validation passed okay, then it means Etcher wrote every single byte in the image to the drive correctly, so something else must be going wrong.
I'm now downloading `alpine-standard-3.5.2-x86_64.iso` to try flashing it with Etcher myself...

lurch, 2017-05-17:
Yup, I flashed `alpine-standard-3.5.2-x86_64.iso` to a microSD card in a USB SD reader with Etcher (it's the smallest disk-image I've ever written with Etcher, so it went amazingly fast :smiley: ), and when I rebooted my laptop and told the BIOS to boot from the USB device, Alpine Linux loaded up and worked just fine.

Nokia808, 2017-05-17:
Hi. Thank you !

1) do Alpine Linux ask you for localhost name & it's password ? 
I copy image on CD & also same thing: ask me for them !!

2) I restore my USB using Windows 7. Certainly Linux Fedora is better for this but I'm affraid from dd command to erase my hard wrongly since I'm of bad luck.

But in Windows it need further steps (please add them to your page):

"Note: if you received error message after step 8 saying “Windows can not format this drive”, then you need to use disk management as following:
- start the disk management tool by click the "Start" button
- type "diskmgmt.msc" into the search box and press enter
- click on “diskmgmt”
- go to “Disk 1”
- left click on “unallocated space” of “Disk 1”
- then right click on “unallocated space” of “Disk 1”
- from list of “right click” select “New simple volume” & follow wizard (be sure to chose maximum possible capacity in this process)."

Best.

lurch, 2017-05-17:
> do Alpine Linux ask you for localhost name & it's password ?

At a guess, I just logged in as `root` and it logged me in straight away, without asking for a password. If you need more help here, I suggest you try asking at https://www.alpinelinux.org/community/  ;-)

> But in Windows it need further steps

Are you saying that the instructions [here](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#windows) don't work for you? What error message do you get?

> Note: if you received error message after step 8

I'm confused - where does "step 8" come from??

Nokia808, 2017-05-17:
Hi again !

1) [ I suggest you try asking at https://www.alpinelinux.org/community/ ]
I'm very sorry ! I did not notice that Alpine Linux behave in this way ! It has no GUI for installation. I discovered how to install it:
https://wiki.alpinelinux.org/wiki/Install_to_disk

2) However, this does not change the fact that Alpine need to be supported by Etcher. Currently Etcher seem to install Alpine (copy it) to USB correctly (please issuer this I'm not sure), but this USB then will become after installation resistant to format & need to be dealt to delete it's file system & recover it. Please your kind fix for this problem.
By the way, I'm already enable validation option & did not receive from it an error message !

3) [ I'm confused - where does "step 8" come from?? ]

Sorry, I dictates all what I learn to documents & save them. I was fast so copy past quickly. Take my explanation:
I applied what existing in your page as following:

1. Open Run ( Windows + R ).

2. Type Diskpart and press Enter ( diskpart ).

3. Type list disk ( list disk ).

4. Type select disk 1 ( select disk 1 ).

5. Type clean and enter ( clean ).

6. Now you USB will show 0Bytes. Don't panic. Just unplug USB and insert again.

7. The system will prompt for fromat and do format.

That’s it. You have recovered Your USB drive to its full capacity from corruption!!

After applying step 7 & got error message say "Windows can not format this device". So, I deal with this by:

**Note: if you received error message after step 7 saying “Windows can not format this drive”, then you need to use disk management as following:
- start the disk management tool by click the "Start" button
- type "diskmgmt.msc" into the search box and press enter
- click on “diskmgmt”
- go to “Disk 1”
- left click on “unallocated space” of “Disk 1”
- then right click on “unallocated space” of “Disk 1”
- from list of “right click” select “New simple volume” & follow wizard (be sure to chose maximum possible capacity in this process).**

It was old document that I used it since long time & your page instruction contain what = to 1st 7 steps. And this remained me by my old document which contain remaining steps.

It was "if you receive error message after step 7 ...... " not step 8.

It will be useful if you add what I put in BOLD to your page.

lurch, 2017-05-17:
> However, this does not change the fact that Alpine need to be supported by Etcher.

Etcher *does* 'support' AlpineLinux - it just copies the exact bytes from the disk image onto the USB drive, exactly as it does for any other disk image. I believe your problems are due to AlpineLinux using a "hybrid ISO" which allows the same image to be booted from both a CD and a USB stick - see https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos for more info.

> By the way, I'm already enable validation option & did not receive from it an error message !

That's good, it means Etcher has done it's job correctly! :-)

> After applying step 7 & got error message say "Windows can not format this device".

@jviotti / @jhermsmeier Are you able to reproduce this problem with our existing Windows instructions not being able to 'recover' a drive that's had AlpineLinux flashed to it?

jviotti, 2017-05-17:
Not really, but I experienced similar issues in the past with certain drives (particularly low quality ones). It'd be nice if @Nokia808 can paste the exact error message and/or the Event Log entry, but I believe that if you keep running `clean` for a couple of times `diskpart` will eventually succeed (like we do on the writer.)

closed 2017-06-16 by jviotti

tiago0289, 2017-08-08:
I am having the same problems, but I already used diskpart, gparted, dd command, sd foematted and nothing can format my pendrive. Before I had 32GB after I used Etcher shows 31MB and in the windows disk manager there is neither an unallocated screen nor other RAW formatted partitions. Only a 31MB RAW partition. Can anybody help me? I fell for my English.

lurch, 2017-08-08:
@tiago0289 Can you download and run the `linux.sh` script from [here](https://github.com/resin-io-modules/drivelist/tree/master/scripts) on a Linux computer with your pendrive inserted, and let us know what it says? Thanks.
(it seems very odd that even `dd` doesn't 'reset' your drive capacity properly. Can you provide a weblink to a product page for your pendrive?)

tiago0289, 2017-08-08:
Good evening. I ran the script and the result was that:
`device: /dev/sde
displayName: /dev/sde
description: "SD/MMC"
size: 32096256
mountpoints: []
raw: /dev/sde
protected: False
system: False
`
O.S: Arch Linux
USB drive: https://www.sandisk.com/about/media-center/media-resources/product/retail/mobile-memory/sandisk-ultra-dual-usb-drive

lurch, 2017-08-09:
Yup, so even `drivelist` (which in turn uses `lsblk`) thinks your drive has a raw capacity of only 31 MB.
That's really weird, and I don't know what could cause that :-/  Are you sure your USB pendrive is a genuine 32GB drive, and not a possibly-fake 31MB drive which has been fraudulently labelled as 32GB? Depending on how recently you bought it, it might be worth trying to get a replacement?

tiago0289, 2017-08-09:
I bought this pendrive for 2 years in a great network of famous Verejistas here in Brazil. I'm sure it's 32GB because I've put that amount of data in it and managed to use it normally. I opened RMA in sandisk, but I'm sure the problem was etcher, I already had the same problem with LiLi USB Creator, but with another model from another brand, another O.S ... thanks for the attention

_Sent from my Sony E6683 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

lurch, 2017-08-09:
One of the Etcher developers also lives in South America... :)  If you're unable to get anywhere with your RMA process, would you be willing to post your USB drive to @jviotti for further investigation?

tiago0289, 2017-08-09:
Of course. I will wait for the answer from sandisk and then try to contact to send the pendrive. From what I've been researching, it looks like the MBR partition is corrupted or something, I'll put the data together and post it later.

_Sent from my Sony E6683 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

lurch, 2017-08-10:
> it looks like the MBR partition is corrupted

Even if the MBR _is_ corrupt, the `lsblk` command (used by `drivelist`) should still report the correct raw capacity. And the `dd` command on https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#gnulinux should wipe out the MBR anyway, allowing standard formatting tools to work properly again.

Fractal7, 2017-08-22:
I've run into a similar issue using etcher for my raspian-jesse-lite image.
The SDcard I was using was an 8Gb, 7$ Sandisk. However after usage with etcher the partition would shrink to 41MB leaving the rest of the unallocated space unusable. Windows formatting did not fix/restore my SDcard to allow 8Gb once again. However after using "Tuxera SD Card formatter" It restored my SDcard and fixed the partition table. 

Thinking that maybe the issue is that it was a cheap sdcard I just tried 64Gb card, one I had previously used to store photos on my DLSR camera and ran into the same issue.

jviotti, 2017-08-22:
@Fractal7 Etcher copies images byte by byte to the drive, without making any modifications, which means that the partition table that your drive acquires is the partition table of the image you flash to it. If `raspbian-jessie-lite` decides that its first partition is 41 MB, then that's what you will see on your drive, independently on its size.

What operating system are you running? Keep in mind that Windows can only detect the first partition of a removable device, so the remaining partitions might be there already, just that Windows can't see them.

The best tool I found for wiping up a partition table is `diskpart`. You can do `diskpart.exe`, `select disk N` (where N is the corresponding device number), and then `clean`.