balena-io/etcher #1443: Feature request: update notifications

TotalCaesar659, 2017-05-17:
Hello! Could you update the notifications? Here is example of Transmission:
[image]
and here is Etcher
[image]
It can look more pretty with the icon https://github.com/resin-io/etcher/blob/master/assets/icon.png and name of ISO image.

lurch, 2017-05-17:
Nice suggestion, thanks :)

jviotti, 2017-05-17:
Cool, let's go for it!

closed 2017-05-22 by jviotti

lurch, 2017-05-22:
@TotalCaesar659 You can test this out in the latest snapshot build of Etcher:
https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-05-22/1.0.0%2B3df6cd0/Etcher-1.0.0%2B3df6cd0-linux-x64.zip

TotalCaesar659, 2017-05-24:
There is no icon in notification with this build.
[image]

lurch, 2017-05-24:
@Shou ?

Shou, 2017-05-25:
I see what's happened, relative path adding some confusion here and pointed to a non-existent icon after some changes. I've submitted a fix.

lurch, 2017-06-01:
@TotalCaesar659 https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-06-01/1.0.0%2Bee3c146/Etcher-1.0.0%2Bee3c146-linux-x64.zip should now fix the missing icon issue, if you want to test the new version :)

TotalCaesar659, 2017-06-02:
Yeah, it looks awesome, thanks :-)
[image]

alexandrosm, 2017-06-02:
I'm concerned about that white line at the bottom of the window, though. Is
that standard for all windows in that os?

jviotti, 2017-06-02:
That while line is coming from the window manager the user is running on his system.

—
Juan Cruz Viotti

--- original message ---

—

--- end of original message ---

TotalCaesar659, 2017-06-03:
@alexandrosm It's the taskbar from window manager LXDE that is used on Lubuntu, called LXPanel.