balena-io/etcher #1444: Linux x86 CLI doesn't work!

lurch, 2017-05-17:
I was experimenting in some LiveCD VM images this morning to check that the 'x64' Linux programs work on a 64-bit host, and the 'x86' Linux programs work on a 32-bit host.

Turns out that the `Etcher-cli-1.0.0-linux-x64.tar.gz` is actually missing some required files :-(

The x64 CLI works on a 64-bit host, and has the following layout:
```
$ tree Etcher-cli-1.0.0-linux-x64/
Etcher-cli-1.0.0-linux-x64/
├── etcher
├── index.js
└── node_modules
    ├── lzma-native
    │   ├── binding
    │   │   └── lzma_native.node
    │   ├── build
    │   │   └── Release
    │   │       ├── lzma_native.node
    │   │       └── obj.target
    │   │           └── lzma_native.node
    │   └── deps
    │       ├── bin_i686
    │       │   └── liblzma.dll
    │       └── bin_x86-64
    │           └── liblzma.dll
    └── mountutils
        └── build
            └── Release
                ├── MountUtils.node
                └── obj.target
                    └── MountUtils.node

13 directories, 9 files
```

However the x86 CLI errors with:
```
module.js:457
    throw err;
    ^

Error: Cannot find module '/path/to/Etcher-cli-1.0.0-linux-x86/node_modules/lzma-native/binding/lzma_native.node'
    at Function.Module._resolveFilename (module.js:455:15)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
[snip]
```
and has the following directory layout:
```
$ tree Etcher-cli-1.0.0-linux-x86/
Etcher-cli-1.0.0-linux-x86/
├── etcher
├── index.js
└── node_modules
    ├── lzma-native
    │   └── deps
    │       ├── bin_i686
    │       │   └── liblzma.dll
    │       └── bin_x86-64
    │           └── liblzma.dll
    └── mountutils
        └── build
            └── Release
                ├── MountUtils.node
                └── obj.target
                    └── MountUtils.node

9 directories, 6 files
```

BTW, I guess there's no need for the Linux CLI archives to contain the Windows `.dll`s? ;-)  And perhaps some of the duplicate `*.node` files can be removed too?

closed 2017-05-17 by jviotti

lurch, 2017-05-18:
Thanks for the rapid fix @jviotti :1st_place_medal: 

Please re-ping this issue when the 'fixed' Linux x86 CLI is available on https://etcher.io/cli

jviotti, 2017-05-18:
@luch I just re-uploaded x86. It should all be good now.

lurch, 2017-05-18:
Yup, just tested in a x86 LiveCD VirtualBox, and the Linux x86 CLI seems to work fine now :+1: 
BTW you pinged the wrong Github user on your last comment ;-)