balena-io/etcher #1450: Etcher CLI looks for index.js in pwd instead of relative to path of etcher binary

tkfu, 2017-05-18:
- **Etcher version:** CLI 1.0.0
- **Operating system and architecture:** Ubuntu 16.04 x64
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** n/a

Like the title says. Some terminal output illustrating the problem:

```
jon@jon-lenovo ~ $ etcher -v
module.js:457
    throw err;
    ^

Error: Cannot find module '/home/jon/index.js'
    at Function.Module._resolveFilename (module.js:455:15)
    at Function.Module._load (module.js:403:25)
    at Module.runMain (module.js:590:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
jon@jon-lenovo ~ $ mkdir -p some/random/dir
jon@jon-lenovo ~ $ cd !!:2
cd some/random/dir
jon@jon-lenovo ~/some/random/dir $ etcher -v
module.js:457
    throw err;
    ^

Error: Cannot find module '/home/jon/some/random/dir/index.js'
    at Function.Module._resolveFilename (module.js:455:15)
    at Function.Module._load (module.js:403:25)
    at Module.runMain (module.js:590:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
jon@jon-lenovo ~/some/random/dir $ cd ~/bin
jon@jon-lenovo ~/bin $ etcher -v
1.0.0
```

Note: I put the contents of the tarball in `~/bin` instead of `/opt/etcher-cli` as suggested, but I also tested it with the contents in `/opt/etcher-cli` just in case there were some hard-coded path searching shenanigans going on. Same problem.

tkfu, 2017-05-18:
It looks like it works if I explicitly specify the path:
```sh
jon@jon-lenovo ~ $ ~/bin/etcher -v
1.0.0
```
This is kinda crappy behaviour, though, and it's undocumented to boot.

lurch, 2017-05-18:
Yep, I can confirm this behaviour.

> This is kinda crappy behaviour

To be fair, we did state that this was the first experimental release of the CLI ;-)
I'm afraid that the CLI packaging took us quite a while to sort out, and it was only completed shortly before the release of Etcher 1.0.0, so it didn't get as much testing as we'd ideally have liked.
Thanks for the bug report!

> and it's undocumented to boot.

@jviotti should we update https://github.com/resin-io/etcher/blob/master/docs/CLI-INSTALLATION.md as a temporary measure until after this is fixed?
Is this isolated to Linux or does the same bug occur on OSX and Windows too?

tkfu, 2017-05-18:
Just tested it on OSX; same problem. I don't have any Windows machines around, though.

closed 2017-07-11 by jviotti