balena-io/etcher #1451: Feature request: option to resize partitions/filesystems on disk to fill available space

tkfu, 2017-05-18:
- **Etcher version:** All
- **Operating system and architecture:** All
- **Image flashed:** Raw disk images
- **Do you see any meaningful error information in the DevTools?** n/a

Use case for this feature request: I have a build process whose output artefact is a raw disk image with two partitions. This image then gets flashed onto boot media of various sizes. Since the size isn't known ahead of time, the image can't be pre-sized to the boot media. That means the process for flashing looks like this, resizing the partition with parted and then the filesystem with e2fstools:

```sh
etcher my-image.img # or dd if=my-image.img of=/dev/sdX if you're feeling dangerous
parted -s /dev/sdX resizepart 2 '100%'
resize2fs /dev/sdX2
```

Could this be implemented as part of Etcher? For the UI, perhaps a check-box for 'Attempt to maximize partition size', and a similar command line option?

lurch, 2017-05-18:
I'm fairly sure we discussed this before somewhere, but Github search isn't finding the right issue discussion for me...

Etcher simply copies the disk image byte-perfect onto the destination drive, and doesn't mess about with the partition tables at all. So I dunno if we'll be adding this feature anytime soon.
In the meantime, perhaps you could use the same workaround that Raspbian uses, where it automatically enlarges the root partition the first time it boots up?

probonopd, 2017-12-02:
I would also be interested in seeing this.

lurch, 2017-12-02:
> I'm fairly sure we discussed this before somewhere...

Aha, here's the other place where it was being discussed: https://github.com/resin-io/etcher/issues/413#issuecomment-263056113

GuBi34, 2018-10-08:
I considder this to be a usefull feature as well (and would like to see it?)

daghemo, 2019-01-15:
Would be a(nother) possibile workaround to https://github.com/balena-os/balena-os/issues/396.

bcomnes, 2019-02-04:
This is surprisingly a pain in the butt on macOS, etcher having this feature would be fantastic, as my SD cards rarely match the image size that comes with the image.

scottrpaterson, 2020-09-07:
This would be an amazing feature to add!

bcomnes, 2020-09-07:
I ended up finding a way to set this up outside of balena by modifying images with packer. 

Script that does it: https://github.com/bcomnes/raspi-packer/blob/master/resizerootfs

Lifecycle script that runs it (systemd): https://github.com/bcomnes/raspi-packer/blob/master/resizerootfs.service

cyberb, 2021-02-14:
Syncloud project is willing to pay for this feature £100 even if it is a Linux only option, job proposal is published at upwork.com