balena-io/etcher #1453: SD Card not getting unmounted in Windows 10 64 bit

enricodetoma, 2017-05-18:
- **Etcher version:** 1.0.0 x64
- **Operating system and architecture:** Windows 10 Home 64 bit version 1607 build 14393.1198
- **Do you see any meaningful error information in the DevTools?** No.

After flashing the image, Etcher is stuck at this screen:

[image]

jviotti, 2017-05-18:
Hi @enricodetoma,

Thanks for reaching out. Can you open DevTools by pressing Ctrl+Alt+I and paste what you see there? There should be some `mountutils` related logging that would be very beneficial to investigate what's going on.

enricodetoma, 2017-05-18:
I tried again with the same image, and now it unmounted successfully.
Is there a way to get the old DevTools log somewhere?

lurch, 2017-05-18:
Nope, it's just like in your web-browser; the console output gets discarded every time you close the program (it's only stored in memory).

lurch, 2017-05-18:
But please do let us know if you find a way to reproduce the issue!

wblondel, 2017-05-18:
Hi,

I have the same problem.

However I don't see any problems in the log file.
[-1495125185366.txt](https://github.com/resin-io/etcher/files/1011600/-1495125185366.txt)

lurch, 2017-05-18:
@wblondel It looks like you're still using 1.0.0-rc.5 - could you please update to 1.0.0 and then post the DevTools log from that if you still get the problem? Thanks.

wblondel, 2017-05-18:
@lurch I can't find the installer for 1.0.0... only the CLI. Where is it? Thanks

lurch, 2017-05-18:
https://etcher.io/ :-)
(scroll down to the bottom if it doesn't offer you the right download for your OS)

wblondel, 2017-05-18:
Oh now I can find it. I swear 10 minutes ago there was only the CLI option aha

jviotti, 2017-05-18:
Hm, so mountutils is not even getting called. Could it be an issue when loading the native module? What OS are you running @wblondel ?

wblondel, 2017-05-18:
I'm using Windows 10 x64 v. 1703.

I can't try again with the newest version of Etcher, because the  whole Windows Explorer freezes when I put my µSD in my µSD reader........

wblondel, 2017-05-18:
Ok I tested with the latest version of Etcher, it works :) The "Unmounting..." part is a bit long (almost 2 minutes), but it works.

The Windows Explorer is still unusable afterwards, I'm obliged to force-reset my computer, but I guess it's just a feature of WinShit 10 =)

[-1495129149827.txt](https://github.com/resin-io/etcher/files/1011790/-1495129149827.txt)

jviotti, 2017-05-18:
> The "Unmounting..." part is a bit long (almost 2 minutes), but it works.

That's definitely not normal. I wonder if the issue both of you are experiencing is related to some Windows glitch.

wblondel, 2017-05-18:
On Linux (Manjaro) everything works fine and the "unmounting..." part is very fast :) And the OS doesn't freeze

enricodetoma, 2017-05-23:
I tried again with a different image, but I was unable to reproduce the problem. Should I close this issue?

jviotti, 2017-05-23:
@wblondel does the issue still happen for you?

wblondel, 2017-05-24:
As I said the unmounting part works but lasts ~2 minutes on Windows.

jviotti, 2017-05-24:
OK, cool. Let's keep this one open then. I'm focusing on the Etcher CLI, but I'll come back to this as soon as I can.

Larswad, 2017-08-18:
For me it never ended. It got stuck at unmounting forever. (Win10 x64).
It's possible to close the application even though its kind of annoying.
Love the Etcher otherwise because it supports compressed formats, big plus.
One thing to mentiond while I'm here (don't have time to create new issue):

* Think absolutely validation should be off by default, SD card failures are very rare.

jviotti, 2017-08-21:
@Larswad I'll keep investigating to see if I can manage to reproduce and fix this issue.

> Think absolutely validation should be off by default, SD card failures are very rare.

We collect various anonymous stats, and you won't believe the large amount of SD card failures out there :)

dj4ngo, 2017-09-12:
HI I'm facing the same issue on Linux Ubuntu 17.04 x86_64. 
I'm using etcher-1.1.2-x86_64.AppImage. 
This  is the content of my ctrl+alt+I console output (before trying to relaunch since it seems to happen randomly)
[etcher.log.zip](https://github.com/resin-io/etcher/files/1297595/etcher.log.zip)
I didn't create new issue related to Linux... I could if you want ... 
I hope it will help you cause I don't see anything suspect inside ...

dj4ngo, 2017-09-12:
It finally finished well after running during about 1 hour ... I was installing raspbian strech lite image ...

Kasey13, 2017-09-24:
I am on Windows 7 64bit and I get unmount failed on every write I try. What is this DevTools Console? I try CTRL+ALT+I and nothing happens. I am using Etcher 1.1.2 64bit version.

petrstetina, 2017-10-25:
Hi, I am facing the same issue - last etcher 64 bit on Gentoo showing Unmounting for hours.
Tested with few different Micro SDHC cards. 

Console does not show any interesting output but ps and strace do.
```# ps -A | grep etcher
 7110 pts/3    00:00:46 etcher-electron
 7113 ?        00:00:06 etcher-1.1.2-x8
 7145 pts/3    00:00:00 etcher-electron
 7164 pts/3    00:01:43 etcher-electron
 7170 pts/3    00:01:49 etcher-electron
 7194 pts/3    00:00:00 etcher-electron
```
As you can expect pid 7113 is waiting for mutex:
`futex(0x7fffb5352ac0, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, 0xffffffff`

Other threads are also waiting for each other but thread 7110 is trying to read some already closed file...

closed 2018-10-15 by jviotti