balena-io/etcher #1457: Feature request(s): Differentiate bw Flash/SD Card & Remember last device

HarikalarKutusu, 2017-05-18:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** Win x64 - but may be in all
- **Image flashed:** Any (Armbian / Raspbian etc)
- **Do you see any meaningful error information in the DevTools?** N/A

As a developer I have multiple USB Flash + SD Card readers. One of the flash drives keeps my SSH key info and 99% already plugged in, so it comes first. First run is OK, but after clicking "flash another" the first device in the list becomes selected - as SD card ejected, it is not on the list. I once cleaned my key files :(

I can think of some possible remedies:
1) Do not auto-select device even there is a single one.
2) Remember the last device written to and select it
3) Put two radio buttons on the main screen: Prefer Flash, Prefer SD Card
4) Have an entry in "settings/preferences" to do the same...

lurch, 2017-05-18:
> I once cleaned my key files :(

Sorry to hear that - I hope you had a backup! I guess if you were storing them on a SD card, you could have flipped the write-protect switch?

> Do not auto-select device even there is a single one.

I think that'd be a backwards step UX-wise, and would annoy many users.

> Remember the last device written to and select it

I'm not sure if this makes sense, especially as on some OSes the identifiers given to different drives depends on the order in which they're plugged in (i.e. plugging in the drives in a different order might give them different identifiers).

> Put two radio buttons on the main screen: Prefer Flash, Prefer SD Card

Unfortunately with some USB card readers (there's *many* different types!), all we know is that it's a USB Mass-Storage-Device - we can't tell if it's SD, Flash, HDD, SSD, etc.

If it's always the same drive that you want to write to, and you're _sure_ that it always appears on your system with the identical identifier, maybe the Etcher CLI (where you can specify the drive as a parameter) would be more suitable for you?

HarikalarKutusu, 2017-05-18:
Yep I had a backup but until I realized what happened I was kept away from my servers :/

I didn't look at CLI version yet (as it is experimental) but it looks promising. I'm assigning the devices into solid drive letters (Q: R: S: etc) and I was already using EjectMedia from Drive Tools (http://www.uwe-sieber.de) as shortcuts...

On the other hand with so many images and drives in hand it would not be feasible... I need to write a tool to select the image interactively (hmmm I had a similar thing in a BAT file).

People ( i.e. I ) must be more careful...

lurch, 2017-05-18:
In the future we're hoping to release an Etcher SDK, which will make building such custom tools much easier :-)

In the meantime, it looks like https://www.google.co.uk/search?q=batch+file+select+file might help you achieve what you're after?

closed 2018-10-15 by jviotti