balena-io/etcher #1458: Error: You don't have access to this resource

HarikalarKutusu, 2017-05-18:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** Win 10 x64
- **Image flashed:** Any
- **Do you see any meaningful error information in the DevTools?** N/A

When etcher started it gives an error but works as intended. Currently using Etcher-1.0.0-win32-x86.exe installation.

> Error
> You don't have access to this resource
> Please ensure you have necessary permissions to access this resource.

Tried and no change:
Elevated run
Disabling Antivirus (Comodo)
Used 2 other portable versions

lurch, 2017-05-18:
Why are you running the x86 (32-bit) version of Etcher on Win 10 x64 (64-bit) ?
Do you get the same result using the 64-bit version of Etcher?

What do you mean by "gives an error but works as intended" ? Can you please post the DevTools console output? (press `Ctrl+Alt+I` )  A screenshot of the error window might be helpful too.

Has your Windows Script Host been disabled? ( https://github.com/resin-io/etcher/issues/1054#issuecomment-296177933 )

HarikalarKutusu, 2017-05-19:
First of all only x86 version was available at downloads as an installer ON THE TOP PULL-DOWN. I installed it... I just realized there is a download section below which also includes 64 bit installer "Etcher-1.0.0-win32-x64.exe" ... (Normally 32 bit versions do work on newer 64 bit windows systems.)

The 32 bit version was installed in: C:\Program Files (x86)\Etcher
I uninstalled 32 bit version, installed 64 bit version (Etcher-1.0.0-win32-x64.exe). 
The 64 bit version is also installed in: C:\Program Files (x86)\Etcher
It gives the same error

I found the error reason in Developer Tools Console tab. It was a firewall issue... See below...

Answers to other questions:

The error pop-up is as follows:

[image]

With "gives an error but works as intended" I meant: Although there was an error shown at the very start, after pressing OK, the program started and worked fully as it should (at least in my 10+ uses)...

I alread did try the portable versions. Result: the same error.

Developer Tools - Console contents:
`C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 document not ready yet, trying again in 500 milliseconds...
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Fri May 19 2017 18:46:30 GMT+0300 (Turkey Standard Time) Application start
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Fri May 19 2017 18:46:30 GMT+0300 (Turkey Standard Time) Checking for updates ({"currentVersion":"1.0.0","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0"})
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 {"cause":{"code":"EACCES","errno":"EACCES","syscall":"connect","address":"52.216.1.216","port":443},"isOperational":true,"code":"EACCES","errno":"EACCES","syscall":"connect","address":"52.216.1.216","port":443}
console.(anonymous function) @ C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51
http://api.mixpanel.com/track/?data=eyJldmVudCI6ICJBcHBsaWNhdGlvbiBzdGFydCI…TYzZTAwOTI4ZGE2N2QxMjI2MzY0ZGQ0YyIsIiRfX2MiOiAwfX0%3D&ip=1&_=1495208790506 Failed to load resource: net::ERR_NETWORK_ACCESS_DENIED
http://api.mixpanel.com/track/?data=eyJldmVudCI6ICJDaGVja2luZyBmb3IgdXBkYXR…jNDU2M2UwMDkyOGRhNjdkMTIyNjM2NGRkNGMiLCIkX19jIjogMH19&ip=1&_=1495208790507 Failed to load resource: net::ERR_NETWORK_ACCESS_DENIED
http://api.mixpanel.com/decide/?verbose=1&version=1&lib=web&token=63e5fc4563e00928da67d1226364dd4c&ip=1&_=1495208790692 Failed to load resource: net::ERR_NETWORK_ACCESS_DENIED
http://api.mixpanel.com/track/?data=eyJldmVudCI6ICJtcF9wYWdlX3ZpZXciLCJwcm9…uIjogIjYzZTVmYzQ1NjNlMDA5MjhkYTY3ZDEyMjYzNjRkZDRjIn19&ip=1&_=1495208790059 Failed to load resource: net::ERR_NETWORK_ACCESS_DENIED
https://sentry.io/api/135211/store/?sentry_version=7&sentry_client=raven-js%2F3.14.1&sentry_key=b073510bafcb4aeeb67aa9d8afa26e7d Failed to load resource: net::ERR_NETWORK_ACCESS_DENIED
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Fri May 19 2017 18:48:00 GMT+0300 (Turkey Standard Time) Navigate ({"to":"settings","from":"main"})
http://api.mixpanel.com/track/?data=eyJldmVudCI6ICJOYXZpZ2F0ZSIsInByb3BlcnR…TYzZTAwOTI4ZGE2N2QxMjI2MzY0ZGQ0YyIsIiRfX2MiOiAyfX0%3D&ip=1&_=1495208880591 Failed to load resource: net::ERR_NETWORK_ACCESS_DENIED
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Fri May 19 2017 18:48:14 GMT+0300 (Turkey Standard Time) Navigate ({"to":"main","from":"settings"})
http://api.mixpanel.com/track/?data=eyJldmVudCI6ICJOYXZpZ2F0ZSIsInByb3BlcnR…jNDU2M2UwMDkyOGRhNjdkMTIyNjM2NGRkNGMiLCIkX19jIjogMX19&ip=1&_=1495208894340 Failed to load resource: net::ERR_NETWORK_ACCESS_DENIED`

I use a firewall which prevents any connection outside by default (as any firewall has to be). Now I see lots of resources need to be downloaded from web. After opening the firewall it worked.

I never though a sd card / flash image writer has to connect to Internet... I'm afraid there will be similar issues among users in the future... I also could not find a proper documentation about this... 
Disabling the errors/usage statistics option in settings also does not solve the problem.
You may like to indicate that "Internet access is required".

Maybe including all resources in the package and checking the network condition and failing silently will solve this problem...

Thank you for your support.

lurch, 2017-05-19:
Aha, thanks for the additional info and well done for tracking down the cause.
I _would_ expect "Disabling the errors/usage statistics option in settings" to solve the problem though, I guess we'll need to do some additional testing... could you provide details about what firewall you're using please?

> You may like to indicate that "Internet access is required".

Etcher _does_ work fine without an internet connection, I guess you've just discovered an edge-case that we hadn't considered.

lurch, 2017-05-19:
> The 64 bit version is also installed in: C:\Program Files (x86)\Etcher

Yeah, that's something we've noticed before. Etcher x64 itself _is_ 64-bit, but IIRC it ends up in `Program Files (x86)` because the installer we use is 32-bit.

HarikalarKutusu, 2017-05-19:
I think so :)

I have several layers of firewalls in the company, but this particular time it was the simple/local "Windows 10 Firewall Control"... No more info from me about our firewalls :D

http://www.sphinx-soft.com/de/Vista/index.html

But in any firewall set as "do not dial out" there will be a problem...

Keep up your good work and thank you for this great piece of software...

jviotti, 2017-05-19:
Looks like the issue is caused when trying to fetch the S3 packages list. Adding a link EACCES catch block at https://github.com/resin-io/etcher/blob/master/lib/shared/s3-packages.js#L150 should do the trick.

closed 2017-05-20 by jviotti

lurch, 2017-05-20:
@HarikalarKutusu If you want to give https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-05-20/1.0.0%2B705e273/Etcher-1.0.0%2B705e273-win32-x64.exe a test, hopefully you'll find that the problem is now fixed?

jviotti, 2017-05-21:
Snapshot builds in action! :P

HarikalarKutusu, 2017-05-21:
I confirm the problem is fixed the version given.

Thank you :)

lurch, 2017-05-22:
@jviotti Yeah, occurred to me that it kind of fits with @alexandrosm 's "full process circle" vision :-)