balena-io/etcher #1459: Error: File path or name contains unicode character

jiejiecn, 2017-05-19:
- **Etcher version: 1.0.0**
- **Operating system and architecture: WIN10 x64 Chinese**

If the image file is located in a unicode path or file name, Etcher rasied an error "Unable to access file"
The image xxxxxxxxxx is not accessible

[image]

lurch, 2017-05-19:
Ahhh, thanks for the bug report @jiejiecn , looks like I also need to add Unicode characters to the testing I'm doing in #1376

lurch, 2017-05-19:
I've done some testing, and this appears to be limited to Windows (hands up all those who find that surprising!).

I created a test directory named `test丠ir` and placed an ISO inside it, and on Linux Etcher 1.0.0 was able to flash the iso okay, but on Windows I got a similar error to @jiejiecn above...
[image]

The interesting thing is that the DevTools console displays the file-path correctly:
```
WRITER: Running: C:\Program Files\Etcher\Etcher.exe C:\Program Files\Etcher\resources\app.asar\lib\child-writer\writer-proxy.js C:\Program Files\Etcher\resources\app.asar C:\Users\IEUser\Downloads\test丠ir\alpine-standard-3.5.2-x86_64.iso --drive \\.\PHYSICALDRIVE1 --unmount --check

C:\Program Files\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Fri May 19 2017 15:35:25 GMT+0100 (GMT Daylight Time) Flash error ({"error":{"description":"The image alpine-standard-3.5.2-x86_64.iso is not accessible","report":false},"image":"C:\\Users\\IEUser\\Downloads\\test丠ir\\alpine-standard-3.5.2-x86_64.iso","drive":{"device":"\\\\.\\PHYSICALDRIVE1","description":"VBOX HARDDISK ATA Device","size":5362882560,"raw":"\\\\.\\PHYSICALDRIVE1","system":true,"protected":false,"mountpoints":[],"name":"\\\\.\\PHYSICALDRIVE1"},"uuid":"64fb2e81-68cb-4b6f-a032-8b11a4ee1b47","unmountOnSuccess":true,"validateWriteOnSuccess":true})
C:\Program Files\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Fri May 19 2017 15:35:25 GMT+0100 (GMT Daylight Time) Open modal ({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
C:\Program Files\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 {"description":"The image C:\\Users\\IEUser\\Downloads\\testä¸ ir\\alpine-standard-3.5.2-x86_64.iso is not accessible","report":false}
console.(anonymous function) @ C:\Program Files\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51
C:\Program Files\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Fri May 19 2017 15:43:40 GMT+0100 (GMT Daylight Time) Modal accepted ({"value":false})
```
so I wonder if the problem might be because we're using `cmd.exe` for the elevation?

https://www.google.co.uk/search?q=cmd.exe+unicode shows that lots of other people have had similar problems too :-/  More fiddling needed to see if I can find a solution...

jviotti, 2017-05-19:
Could powershell help us here? Alternatively, we could investigate if the ShellExecuteEx Windows functions allows to somehow pass environment variables to the child process, and see if we can get rid of the cmd.exe abomination

lurch, 2017-05-20:
> see if we can get rid of the `cmd.exe` abomination

That would be nice ;-)   I'll investigate this more next week... (although presumably using ShellExecuteEx would mean writing another native C++ module?)
Having had a **quick** read through https://msdn.microsoft.com/en-us/library/windows/desktop/bb759784(v=vs.85).aspx there doesn't seem to be anything about setting custom environment variables though...

I'm beginning to wonder if it might be 'easier' (i.e. get rid of a whole bunch of problems) to compile a custom version of electron, where the "special" environment variables we need to set can actually also be set with command-line options? :-/

jviotti, 2017-05-20:
> although presumably using ShellExecuteEx would mean writing another native C++ module?

We're already using `ShellExecuteEx` to perform the elevation: https://github.com/resin-io/etcher/blob/master/src/os/win32/elevate.cpp#L80

> I'm beginning to wonder if it might be 'easier' (i.e. get rid of a whole bunch of problems) to compile a custom version of electron, where the "special" environment variables we need to set can actually also be set with command-line options? :-/

Yeah, that could be a good way to go. I wonder if the Electron team are interested in making this "mode" available as a command line option.

> EDIT: Just found https://msdn.microsoft.com/en-us/library/windows/desktop/ms682653(v=vs.85).aspx but I'm too tired to comprehend it fully ATM ;)

Nice find!

> By default, a child process inherits the environment variables of its parent process. Programs started by the command processor inherit the command processor's environment variables. 

I wonder if this means we can set `ELECTRON_RUN_AS_NODE` in the current process (using `SetEnvironmentVariable`) (which will not affect the main GUI since that variable is only read when Electron starts), and that will automatically get inherited.

> To specify a different environment for a child process, create a new environment block and pass a pointer to it as a parameter to the CreateProcess function.

Sadly we can only perform elevation by using `ShellExecuteEx`. What we could do, however, is make the elevation routine spawn a new process with a custom environment using `CreateProcess`, and then call `ShellExecuteEx` from it.

jviotti, 2017-05-22:
I've been experimenting with this, but turns out that when using the `ShellExecuteEx` `runas` verb (for elevation), environment variables won't be preserved :/ See https://stackoverflow.com/a/42385750.

Alternatively, we could make an intermediary tool that takes environment variables as arguments, sets them, and then executes the writer process.

lurch, 2017-05-22:
> but turns out that when using the ShellExecuteEx runas verb (for elevation), environment variables won't be preserved

I guess that kinda makes sense, since running the app as a different user, it _is_ then inside a different environment.

>  Alternatively, we could make an intermediary tool that takes environment variables as arguments, sets them, and then executes the writer process.

That might be an interesting approach, as I guess that'd then give us much more control and we wouldn't be limited to the awful quoting / escaping mechanisms used by `cmd.exe`  (see https://github.com/resin-io/etcher/pull/1376 for some of the craziness involved when trying to use `cmd.exe`).
Plus it would also give us the added bonus that the UAC dialog would then display the name of our intermediary tool (which we could name whatever we want), rather than the UAC prompt asking the user to elevate "Windows Command Prompt" (which as I've mentioned before, I don't think is great UX).

closed 2017-07-05 by jviotti

lurch, 2017-07-07:
@jiejiecn https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-07-05/1.0.0%2Bfd109b5/Etcher-1.0.0%2Bfd109b5-win32-x64.exe should have fixed this problem, if you'd like to test? :)

jiejiecn, 2017-07-10:
@lurch  With this build, I get this error when I try to select an image file.

[image]

Let me know anything I can help :-)

lurch, 2017-07-10:
My apologies for linking you to a buggy build. Please try this one instead ;-)
https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-07-06/1.0.0%2B1b79d50/Etcher-1.0.0%2B1b79d50-win32-x64.exe

jiejiecn, 2017-07-11:
@lurch  So far, this one works well.