balena-io/etcher #1460: Incorrect "Too small for image" error on Windows 10

rfjakob, 2017-05-19:
Summary: The SD card is 7948206080 bytes, the image is 7948206080 bytes. On Linux, Etcher reports the correct size and flashes the image without issue. On Windows, Etcher reports 7945620480 bytes, and flashing fails. May be a bug in Windows (see the log on the bottom).

```
Image size      7945620480 bytes
SD card size    7945620480 bytes
Windows reports 7948206080 bytes
```

- **Etcher version:**

1.0.0

[image]

- **Operating system and architecture:**

Windows 10

- **Image flashed:**

Image that was copied *from an identical SD card*

- **Do you see any meaningful error information in the DevTools?**

### Linux
```
Thu May 18 2017 16:32:06 GMT+0200 (CEST) Select image ({"path":"/tmp/sdcard.rk3399-q7.img","extension":"img","size":{"original":7948206080,"final":{"estimation":false,"value":7948206080}},"logo":false,"bmap":false})

Thu May 18 2017 16:32:51 GMT+0200 (CEST) Toggle drive ({"drive":{"device":"/dev/sdb","description":"Card Reader","size":7948206080,"mountpoints":[],"raw":"/dev/sdb","protected":false,"system":false,"name":"/dev/sdb"},"previouslySelected":false})
```

### Windows 10
```
Fri May 19 2017 11:05:03 GMT+0200 (W. Europe Daylight Time) Select image ({"path":"C:\\Users\\cad\\Downloads\\sdcard.rk3399-q7.img","extension":"img","size":{"original":7948206080,"final":{"estimation":false,"value":7948206080}},"logo":false,"bmap":false})

Fri May 19 2017 11:05:06 GMT+0200 (W. Europe Daylight Time) Toggle drive ({"drive":{"device":"\\\\.\\PHYSICALDRIVE1","description":"Multiple Card  Reader USB Device","size":7945620480,"raw":"\\\\.\\PHYSICALDRIVE1","system":false,"protected":false,"mountpoints":[],"name":"\\\\.\\PHYSICALDRIVE1"},"previouslySelected":false})
```

This is consistent with what "wmic" reports, but the number is wrong!

```
C:\Users\cad>wmic diskdrive list brief
Caption                           DeviceID            Model                             Partitions  Size
Samsung SSD 850 EVO 250GB         \\.\PHYSICALDRIVE0  Samsung SSD 850 EVO 250GB         3           250056737280
Multiple Card  Reader USB Device  \\.\PHYSICALDRIVE1  Multiple Card  Reader USB Device  1           7945620480
```

rfjakob, 2017-05-19:
Edit: Table was the wrong way round (Windows reports a smaller size). Corrected.

lurch, 2017-05-19:
Duplicate of https://github.com/resin-io-modules/drivelist/issues/142

At some point we're hoping to rewrite `drivelist` as a native C++ module, which will allow us to then use the "correct" (low-level) method of getting the SD card's size. Until then I'm afraid there's no workaround.

closed 2017-05-19 by lurch