balena-io/etcher #1461: Bug: Writing/validating fails if new SD Card adapter inserted in the process

HarikalarKutusu, 2017-05-20:
- **Etcher version:** 1.0.0 (x64 intaller)
- **Operating system and architecture:** Win 10 x64
- **Image flashed:** Any
- **Do you see any meaningful error information in the DevTools?**

I have multiple sd card readers.
When writing an image to drive 1 I plugged another. The current session cancelled with the following error:

[image]

Tried a couple times in different phases:
* "Starting" Phase - no problem.
* "Writing" Phase - ERROR
(here Windows froze and I had to reset the PC - very uncommon for my setup)
* Verify phase - ERROR

After "error", retry button works of course...
DevTools does not show any extra relevant info.

alexandrosm, 2017-05-20:
Wow, this is fascinating! Thanks, with such a clear report, we should be
able to fix this pretty fast.

jviotti, 2017-05-20:
This is really, really weird. I'll give it a go later, but this means that Windows causes ENOENT or UKNOWN for one drive when plugging another, which is non sense :S

lurch, 2017-05-21:
I've been trying to reproduce this on my Windows 7 x64 laptop this evening, and I was unable to. During both the Flashing and the Validating phases, I tried inserting / removing two other USB drives many many times, and I couldn't get this error to trigger.

I wonder if maybe inserting the new USB device is causing a momentary 'power glitch' which causes the drive-being-written to temporarily disappear and reappear (which could cause the error above?)  Or maybe the act of inserting a second drive causes enough vibration on the first USB socket to cause it to temporarily lose connection?
I'm no USB expert, but perhaps the large amount of data that gets written and read over USB during Etcher's writing and validating phases make the USB connection "extra sensitive" to very brief glitches, which would otherwise go un-noticed during typical USB usage? \*shrug\*

@HarikalarKutusu Can you provide more details about the SD card readers you're using? Are you connecting them directly to your computer, or through a USB hub?  (obviously if the latter, then it might be worth trying a different hub, or plugging the card readers directly into your computer rather than via a hub?)

HarikalarKutusu, 2017-05-21:
It is very interesting indeed :(
I could not replicate it either :/

Main SD Card Reader was/is plugged directly into a USB 3 port on the front panel. This is an old working USB 2 drive. 
The second and the third ones (yes, same cheap Chinese USB 3 readers, which have been working fine by themselves with Win32DiskImager) were plugged into an un-powered 4 port USB 3 hub, which is further connected to a back-panel USB 3 port, The hub has port switches (on/off).
This setup is meant to maximize throughput and minimize congestion/intervention.

Today I tried many more combinations **without any errors**... 

Used the HUB again, different readers, different ports
Plugged in directly
Used On/Off switches on the hub
Connected to the second front panel port
Tried another similar hub
etc

Nada...

I'm pretty sure it has something to do with the electrical condition of the hub and/or Windows not coping with signals at that specific time.
I also have lots of single board computers around, they could also have intervened at that time...

I'm so sorry for the wrong report and the time loss I caused. This is not a software bug.
But at least we know now something like this can happen...
If it happens again, I'll be able to analyze more facts...
Cheers :)

Bülent

closed 2017-05-21 by HarikalarKutusu

lurch, 2017-05-22:
Thanks for the update @HarikalarKutusu :+1: