balena-io/etcher #1462: [feature] Allow download of specific ISOs

ShalokShalom, 2017-05-20:
I think, it makes the job easier, to implement the function to recompile Etcher with a download link, so the distributions can create Installers, who create a live device, based on one single app (Etcher). +1

lurch, 2017-05-20:
This is a feature we've already got planned for [Etcher V2](https://resin.io/blog/the-future-of-etcher/) :-)

Closing this as a duplicate of #762

closed 2017-05-20 by lurch

ShalokShalom, 2017-08-12:
@lurch This https://github.com/resin-io/etcher/issues/762 is closed, how to work on it?
Can you reopen it? Thanks

lurch, 2017-08-14:
Easiest option will be to just wait for Etcher V2, as this is something we won't be adding to Etcher V1.
Or you could create your own fork...

ShalokShalom, 2017-08-14:
I am waiting for V2. The question is, why this is closed? I mean, where does it get tracked?

lurch, 2017-08-14:
In our internal (private) Etcher V2 design documents...

ShalokShalom, 2017-08-14:
Ah i see.