balena-io/etcher #1465: Etcher is still sending some usage/tracking data though disabled

schlamar, 2017-05-24:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** Win 7, 64bit

I have statistic sending disabled in settings but etcher is still sending some usage/tracking data via HTTP (`http://api.mixpanel.com/track/?data=xxx`).

jviotti, 2017-05-24:
@schlamar That's weird, thanks for reporting. Can you paste the logs from DevTools (Ctrl+Alt+I), so we can see at which part the events are being sent?

schlamar, 2017-05-24:
Looks like every event is sent, e.g. navigating to settings and back.

    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 document not ready yet, trying again in 500 milliseconds...
    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51  _____ _       _
    |  ___| |     | |
    | |__ | |_ ___| |__   ___ _ __
    |  __|| __/ __| '_ \ / _ \ '__|
    | |___| || (__| | | |  __/ |
    \____/ \__\___|_| |_|\___|_|

    Interested in joining the Etcher team?
    Drop us a line at join+etcher@resin.io
    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Wed May 24 2017 15:29:56 GMT+0200 (Mitteleuropäische Sommerzeit) Application start
    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Wed May 24 2017 15:29:56 GMT+0200 (Mitteleuropäische Sommerzeit) Checking for updates ({"currentVersion":"1.0.0","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0","includeUnstableChannel":false})
    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 {"cause":{"code":"UNABLE_TO_VERIFY_LEAF_SIGNATURE"},"isOperational":true,"code":"UNABLE_TO_VERIFY_LEAF_SIGNATURE"}
    console.(anonymous function) @ C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51
    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Wed May 24 2017 15:30:18 GMT+0200 (Mitteleuropäische Sommerzeit) Navigate ({"to":"settings","from":"main"})
    C:\Apps\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Wed May 24 2017 15:30:23 GMT+0200 (Mitteleuropäische Sommerzeit) Navigate ({"to":"main","from":"settings"})

jviotti, 2017-05-24:
Could you open the DevTools "Application" tab, click `file://` under the "Local Storage" section and show me the stringified JSON contents of the "redux" key? For example:

[image]

```json
{"unmountOnSuccess":true,"errorReporting":false,"sleepUpdateCheck":false,"lastUpdateNotify":1494444978918,"unsafeMode":false,"downloadPath":"/Users/jviotti/Downloads","ejectOnSuccess":true,"validateWriteOnSuccess":false,"useUnstableUpdateChannel":false}
```

That `{"cause":{"code":"UNABLE_TO_VERIFY_LEAF_SIGNATURE"},"isOperational":true,"code":"UNABLE_TO_VERIFY_LEAF_SIGNATURE"}` error is very interesting as well. I'll send a PR for it.

jviotti, 2017-05-24:
Regarding the `UNABLE_TO_VERIFY_LEAF_SIGNATURE` error, are you behind a proxy?

schlamar, 2017-05-24:
Content of "redux":

    {"unsafeMode":false,"errorReporting":false,"unmountOnSuccess":true,"validateWriteOnSuccess":true,"sleepUpdateCheck":false,"includeUnstableUpdateChannel":false,"lastUpdateNotify":null}

> Regarding the UNABLE_TO_VERIFY_LEAF_SIGNATURE error, are you behind a proxy?

Yes

jviotti, 2017-05-26:
Thanks for the information. I'm sending a fix for the `UNABLE_TO_VERIFY_LEAF_SIGNATURE` error, and @Shou can probably help us out getting to the bottom of that ugly analytics issue.

jviotti, 2017-07-17:
I confirm this issue is being fixed by https://github.com/resin-io/etcher/pull/1604

PR #1604 (2017-07-18): upgrade(package): update resin-corvus to 1.0.0-beta.28 
Closes: #1465
closed 2017-07-18 by jviotti