balena-io/etcher #1466: Written Armbian image correctly on 64 GB micro SD but it's not using full available space

lordspace, 2017-05-24:
Hi,

It would have been really nice to have an option to pick how much of the SD card to use or just use it fully.

- **Etcher version:**
Etcher-1.0.0-win32-x64.exe

- **Operating system and architecture:**
windows 8

- **Image flashed:**
https://dl.armbian.com/cubietruck/archive/Armbian_5.25_Cubietruck_Ubuntu_xenial_default_3.4.113_desktop.7z

- **Do you see any meaningful error information in the DevTools?**

```
Filesystem      Size  Used Avail Use% Mounted on
udev            941M     0  941M   0% /dev
tmpfs           189M  5.5M  184M   3% /run
/dev/mmcblk0p1  2.7G  2.2G  490M  82% /
tmpfs           944M   92K  944M   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           944M     0  944M   0% /sys/fs/cgroup
tmpfs           944M  104K  944M   1% /tmp
log2ram          50M  1.4M   49M   3% /var/log
tmpfs           189M     0  189M   0% /run/user/0
tmpfs           189M  8.0K  189M   1% /run/user/1000

```

lurch, 2017-05-24:
Closing as duplicate of #1451

closed 2017-05-24 by lurch

lordspace, 2017-05-24:
Ok. So i formatted the SD card with SD formatter and wrote the image to the SD card again.
Now I am seeing the full disk size ?!?

```
root@cubietruck:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            941M     0  941M   0% /dev
tmpfs           189M  3.2M  186M   2% /run
**/dev/mmcblk0p1   56G  2.4G   53G   5% /**
tmpfs           944M  140K  944M   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           944M     0  944M   0% /sys/fs/cgroup
tmpfs           944M  104K  944M   1% /tmp
log2ram          50M  1.4M   49M   3% /var/log
tmpfs           189M  8.0K  189M   1% /run/user/1000
tmpfs           189M     0  189M   0% /run/user/0

```

lurch, 2017-05-24:
Did you write the image to the SDcard (e.g. using Etcher), or did you just copy the image file into a folder on the SD card? ;)

lordspace, 2017-05-24:
Yep. I wrote it with etcher.
The UI is so nice so it didn't even cross my mind to write it any other way ;)

lurch, 2017-05-24:
Sometimes you need to remove and then re-insert an SD card for the OS to see the updated partition sizes - could that be what's happening here?

lordspace, 2017-05-24:
After I wrote the image for the 2nd time I have just inserted it and haven't touched it since.

ThomasKaiser, 2017-05-25:
This is not an Etcher issue (maybe not even a duplicate): https://forum.armbian.com/index.php?/topic/4359-written-armbian-image-correctly-on-64-gb-micro-sd-but-its-not-using-full-available-space/&do=findComment&comment=32401 (just as a reference for other people stumbling accross this)

lurch, 2017-05-25:
Thanks for the link @ThomasKaiser