balena-io/etcher #1476: Feature Request: Static URL for downloading latest AppImage installer for Linux

teejee2008, 2017-05-26:
Is there a URL for downloading the latest AppImage file for Etcher? I'm looking for a URL that always points to the latest installer so that I can write a script that downloads and installs the latest version without requiring changes in the future.

Something like:
> https://resin-production-downloads.s3.amazonaws.com/etcher/latest/Etcher-latest-linux-x64.zip

If I link to v1.0.0 would the link still work after another version is released?

jviotti, 2017-05-28:
We recently agreed to move to GitHub Releases, which provides this feature out of the box. I'll keep this issue opened until that gets done :)

lurch, 2017-05-31:
Specifically answering...

> If I link to v1.0.0 would the link still work after another version is released?

Yes, we don't delete old versions, so older URLs continue to work. See e.g. the links on https://github.com/resin-io/etcher/releases

jviotti, 2017-05-31:
Yeah, that sounds like a great suggestion.

probonopd, 2017-08-18:
And please one without zip. So that we could get it into https://github.com/AppImage/AppImageHub eventually.

lurch, 2018-02-08:
FYI I've added a new comment to https://github.com/resin-io/etcher-homepage/issues/57

jhermsmeier, 2018-02-26:
Unfortunately GitHub only supports linking to the release itself under a `/latest` URL, i.e. https://github.com/resin-io/etcher/releases/latest – but not to downloads themselves, so we'll need to do some API talking in the website in order to sort those out; but it should certainly be easier now.

closed 2018-10-15 by jviotti