balena-io/etcher #1477: SD Card Backup and .img Files Compression

robertsLando, 2017-05-26:
I mostly use Etcher.io to write Raspberry PI SD cards img and I think there is nothing like Etcher.io, It's the best! I think that most people use Etcher for my same purpose, for this reason I have two important suggestions that can make Etcher.io a MUST HAVE TOOL for Raspberry PI owners.

1.  **SD card backup** (create an .img file of the SD)
2. **Image File Compression**. This is very interesting and there isn't any other software that do this. I'm using [Aoakley](https://github.com/aoakley)'s command line tool called [Raspbian-Shrink](https://github.com/aoakley/cotswoldjam). This tool works just with Raspbian (not NOOBS) but allows to easly and quickly compress img files (my 16GB sd card img was 14.9GB and with this tool --> 1.6GB).

jhermsmeier, 2017-05-26:
Hey @robertsLando, thanks for the suggestions!

1. We're already tracking this as #266 (Implement a way to perform a backup from a drive)
2. Looks interesting, but that actually changes the file systems and partition table of the image, which is technically not really compression, but resizing. This makes sense for Raspbian images as there's an option to expand the file system after booting it, but most other operating systems / distros don't have this option – so I don't think it would make sense for us to put into Etcher (pinging @jviotti for thoughts on this). One technique we do already support (and which can be applied to any image) is blockmapping with [bmaps](https://source.tizen.org/documentation/reference/bmaptool/bmap-tools-project) also allowing much faster flashing by only writing the necessary parts of an image.

jviotti, 2017-05-28:
Hi @robertsLando ,

Thanks a lot for the nice words! We have plans to implement the backup feature soon (hopefully). Regarding the second one, if this is a Raspbian specific thing, then I agree with @jhermsmeier in that its probably outside the scope of this project.

Here's another suggestion (let me know if it addresses your request): when performing a backup from an SD Card, Etcher can read the partition table, and only save what's really needed. For example, if you flashed Raspbian to a 16 GB card, doing the backup will likely only result in ~2 GB or something like that.

robertsLando, 2017-05-28:
@jviotti that is what I mean with image compression. It has happen to me that after the img backup when I try to write the image in the same SD I get the error "img too large to be write" (something like this I don't remember the real message but this is the meaning).

jviotti, 2017-05-29:
Awesome, I'll close this in favor of https://github.com/resin-io/etcher/issues/266

closed 2017-05-29 by jviotti