balena-io/etcher #1483: Ignore Windows drives that report themselves as 0.0GB

jviotti, 2017-06-01:
See https://github.com/resin-io-modules/drivelist/issues/142

In summary, WMI sometimes fails to detect the real size of the drive, and detects a unreasonably small drive size (e.g. 8 MB) that gets coerced into 0.0 GB by the Etcher UI.

We have a real fix for this (which involves to re-write drivelist on C++), but let's make `drivelist` filter these drives for now.

lurch, 2017-06-01:
Umm nope, this is in fact opposite to what @alexandrosm was suggesting in gitter!
We don't want to filter out (ignore) drives that report themselves as 8MB, we want to ignore the size they report, and allow any-size-image to be written to them anyway.

jviotti, 2017-06-01:
Yeah, my bad, I totally mixed things up. I'll rephrase the issue.

alexandrosm, 2017-06-01:
basically we should set a "minimum plausible drive size" and ignore any responses that are below that, simply reporting them as "unknown size", and as a result, not applying any size checks

lurch, 2017-06-01:
...and there's then a nice symmetry between drives who's size we don't know, with images who's size we don't know ;-)

closed 2017-07-12 by jviotti

lurch, 2017-07-13:
Please see https://github.com/resin-io/etcher/pull/1574#issuecomment-315054168

PR #1574 (2017-07-12): upgrade: drivelist to v5.0.25
This version contains various improvements on the drive detection system
mainly on Windows. The improvements can be summarized as follows:

- Drivelist no longer spawns a .bat script to perform its job. This
  means that a lot of spawning related issues are now fixed

- Drivelist doesn't fetch drive sizes from WMI anymore, which was known
  to incorrectly report a very small size for certain drives, causing
  the famous "0.0 GB" Windows issues (see
  resin-io-modules/drivelist#142)

- Cleanup temporary scripts after execution

This PR also sets the DRIVELIST_DEBUG [... cut]