balena-io/etcher #1485: HI i am trying to flash 64gb Micro sd card format is fat32 for a raspberry pi it is not working

MatthewRead42, 2017-06-01:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** 10.10.5 OS X
- **Image flashed:** ubuntu mate 16.04.2 
- **Do you see any meaningful error information in the DevTools?** it as not flashed properly do you want to retry

jviotti, 2017-06-01:
Hi @MatthewRead42 ,

Can you elaborate on why "it is not working"? Do you see an error? If so, at which stage?

MatthewRead42, 2017-06-01:
After it verifies the drive

jviotti, 2017-06-01:
Can you send us a screenshot of the error?

MatthewRead42, 2017-06-01:
[image]
[image]

MatthewRead42, 2017-06-01:
It is a new sd card and i have used a different card to a 256gb and I got the same problem?

jviotti, 2017-06-01:
Can you try another SD Card reader? Etcher will display that error even when a single byte has been written incorrectly, and it may mean that is catching subtle write failures due to either the SD Cards or the reader.

MatthewRead42, 2017-06-01:
ok thanks

jviotti, 2017-06-02:
Keep in mind you can disable validation on the settings page (the cog in the top right corner), but I'd highly advise not do, since there is really something going on that Etcher is catching here.

closed 2017-06-02 by jviotti

Oscarz90, 2018-09-11:
I had the same problem with a sd card of 64gb, I think that it's related with micro sd of big capacity......

lurch, 2018-09-11:
I suspect the bigger the card, the higher the chances of it being a "fake" card?