balena-io/etcher #1486: Use amd64 and i386 as architectures in the build system

jviotti, 2017-06-02:
For consistency with the rest of resin.io.

lurch, 2017-06-02:
...but as I asked in Flowdock, does this also mean changing the filenames that we generate (and offer for download) ?

jviotti, 2017-06-02:
I propose keeping `x64` and `x86` on the final packages, since they are basically node/Electron packages, which follow their own architecture conventions.

lurch, 2017-06-02:
Okay, so I guess we'll need to add `-t outputfile` to `architecture-convert.sh` then ;-)
Ping @alexandrosm for comments as he was the one suggesting this.

lurch, 2018-02-26:
Looks like this has become fairly low priority... :wink: 

For reference, the Etcher [Makefile](https://github.com/resin-io/etcher/blob/master/Makefile) currently deals with these 'canonical' architectures:
 * x64
 * x86
 * armv7hf  (added in #1482 )
 * aarch64  (added in #2042 )

and the [architecture-convert.sh](https://github.com/resin-io/etcher/blob/master/scripts/build/architecture-convert.sh) script then converts these to whatever architecture other systems (`node`, `electron-builder`, `debian`, `redhat`, `appimage` or `docker`) expect.

Similarly, the Makefile also deals with these 'canonical' platforms:
 * win32
 * linux
 * darwin

and the [platform-convert.sh](https://github.com/resin-io/etcher/blob/master/scripts/build/platform-convert.sh) script converts these to whatever platforms other systems (currently just `pkg`) expects.

closed 2018-10-15 by jviotti