balena-io/etcher #1490: fails to start on Linux Mint

jnweiger, 2017-06-04:
- **Etcher version:** Etcher-1.0.0-linux-x86.AppImage 
- **Operating system and architecture:** PRETTY_NAME="Linux Mint 18.1"
- **Image flashed:** none. does not even start
- **Do you see any meaningful error information in the DevTools?** no

After download and unzipping I found the x-bit already set. Appimage instructions say one has to add the x-bit. All the gui starters that I tried failed without an error message.
Trying to start from a console shell yields this error message:

<pre>
1& [testy:~/Downloads] $ ~/bin/Etcher-1.0.0-linux-x86.AppImage 
/home/testy/bin/Etcher-1.0.0-linux-x86.AppImage: error while loading shared libraries: libfuse.so.2: cannot open shared object file: No such file or directory
</pre>

lurch, 2017-06-04:
Does https://github.com/AppImage/AppImageKit/wiki/FUSE help?
Also, are you sure you need to be using the x86 AppImage rather than the x64 AppImage?

> After download and unzipping I found the x-bit already set.

Yeah, that's why we package the AppImage inside a zip, as it allows us to preserve the excecutable (`x`) bit :-)

jviotti, 2017-06-05:
Yeah, I think simply installing `libfuse2` should do the trick.

closed 2017-06-16 by jviotti

matjazmav, 2018-11-09:
I have exact same error, also tried to install libfuse2 but same error.

MacFarlaneBro, 2018-11-10:
I already have libfuse2 installed and I also experienced the same error

sbadux, 2018-11-11:
Same here (Ubuntu 18.10)

matjazmav, 2018-11-11:
Downloaded linux x64 package from https://www.balena.io/etcher/ gave me wrong package.
After I downloaded from github/releses it worked.
[image]

lurch, 2018-11-12:
I wonder if that might be linked to https://github.com/balena-io/etcher-homepage/issues/143 ?

cristenicu, 2018-12-04:
On Ubuntu 18.10 I used the `.deb` file from here https://github.com/balena-io/etcher/releases

buzzdev, 2018-12-07:
> Downloaded linux x64 package from https://www.balena.io/etcher/ gave me wrong package.
> After I downloaded from github/releses it worked.
> [image]

Same here.

MikiDi, 2019-01-27:
Same here as well ...

antony, 2019-02-13:
Can confirm that having `libfuse2` installed does not help on Ubuntu 18.10:

```bash
$ sudo apt search libfuse2
Sorting... Done
Full Text Search... Done
libfuse2/cosmic,now 2.9.8-1ubuntu1 amd64 [installed,automatic]
  Filesystem in Userspace (library)
```

```bash
$ ./balena-etcher-electron-1.4.9-i386.AppImage 
./balena-etcher-electron-1.4.9-i386.AppImage: error while loading shared libraries: libfuse.so.2: cannot open shared object file: No such file or directory
```

```bash
$ cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.10
DISTRIB_CODENAME=cosmic
DISTRIB_DESCRIPTION="Ubuntu 18.10"
NAME="Ubuntu"
VERSION="18.10 (Cosmic Cuttlefish)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.10"
VERSION_ID="18.10"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=cosmic
UBUNTU_CODENAME=cosmic
```

lurch, 2019-02-13:
@antony The **amd64** in `libfuse2/cosmic,now 2.9.8-1ubuntu1 amd64 [installed,automatic]` indicates that you're running a 64-bit version of Ubuntu, but the **i386** in `./balena-etcher-electron-1.4.9-i386.AppImage` indicates you're trying to run the 32-bit Etcher AppImage. The error message about libfuse isn't very helpful, but you can't run 32-bit applications on a 64-bit Linux host.
Try downloading and running the 64-bit AppImage instead :slightly_smiling_face:

donglixp, 2019-09-06:
install libfuse2 solves the problem on ubuntu 16.04

sawaYch, 2020-01-07:
> On Ubuntu 18.10 I used the `.deb` file from here https://github.com/balena-io/etcher/releases

This solves the problem on my machine. I am using KDE NEON (Ubuntu18.04), and fuse2 is already pre-installed.

yung-ber, 2020-08-08:
does anybody have a solution that actually works??
I got the same error. Running LXLE/LXDE linux.

igorgue, 2020-11-03:
I just had this problem right now, I accidentally downloaded the 32 bits one, I think your download selector should show the 64bits version as well, so nobody gets confused thinking only one version exists.

mschaller, 2021-01-07:
> Downloaded linux x64 package from https://www.balena.io/etcher/ gave me wrong package.
> After I downloaded from github/releses it worked.
> [image]

same for me; clicked Linux x64 - got ia32 instead

yung-ber, 2021-01-07:
I think you have mistaken my email for someone else's. Who are you trying
to contact?

sent from my LG phone

yung-ber, 2021-01-07:
Oh wait, sorry about that.

sent from my LG phone

igorgue, 2021-01-07:
But that was my point. The UI is confusing, you think you *have to* click the arrow to open the select but probably you need the "default" which I didn't get at first the control is a "button select" so you don't have to click the arrow to open.

kmanwar89, 2022-06-02:
For anyone else on Ubuntu 22.04 coming across this issue...I had a similar behavior when using Joplin. I followed the instructions [here](https://github.com/AppImage/AppImageKit/wiki/FUSE) for installing the libfuse2 library, which is required for Ubuntu >= 22.04.

liudonghua123, 2022-06-06:
I have the same problems when trying execute 32-bit app packaged in `appimage` format on 64-bit os. 

I remove `fuse3` and install `fuse:i386` instead, then this error disappeared, 32-bit app can run on 64-bit os.

lurch, 2022-06-06:
If you're using a 64-bit OS, it's probably easier to just use the 64-bit AppImage? :shrug: