balena-io/etcher #1492: Etcher losing acces to the drive

brianmituka, 2017-06-06:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** linux mint cinnamon 32bit
- **Image flashed:** raspbian jessie

 **that is the response I get when I try to flash my Sd card**
[image]

**that is the output I get from the developer's console**
`/tmp/.mount_41RM6C/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 06 2017 10:05:58 GMT+0300 (EAT) Drive unplugged ({"image":"/home/mots/Desktop/raspbian/2017-04-10-raspbian-jessie.img","drive":{"device":"/dev/mmcblk0","description":"SL08G","size":7948206080,"mountpoints":[],"raw":"/dev/mmcblk0","protected":false,"system":false,"name":"/dev/mmcblk0"},"uuid":"b01b2a79-6b10-44bb-bdba-65c474fe7c4e","unmountOnSuccess":true,"validateWriteOnSuccess":true})`

**After the failed flashing, this is what I get when I try to open the sd card**
`Error mounting /dev/mmcblk0p1 at /media/mots/4D72-10FE: Command-line `mount -t "vfat" -o "uhelper=udisks2,nodev,nosuid,uid=1000,gid=1000,shortname=mixed,utf8=1,showexec,flush" "/dev/mmcblk0p1" "/media/mots/4D72-10FE"' exited with non-zero exit status 32: mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0p1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
`

lurch, 2017-06-06:
Are you sure your SD reader isn't faulty / flaky, as the error message suggests?
Could you provide more details about which SD reader you're using?

Hmmm, I guess `/dev/mmcblk0` might indicate that it's an internal SD reader slot on a laptop? If so could you provide us with details about your laptop?
Have you ever had any other problems with your SD slot being unreliable with other programs? Do you also have a USB SD adaptor that you're able to test with?

jviotti, 2017-06-06:
Yeah, this really looks like a faulty card, or reader. Its known that these types of issues can trigger a "drive unplugged" event, and `mount` seems to confirm that (`mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0p1`).

I'll close this, since there's probably nothing we can do from our side.

lurch, 2017-06-06:
The `mount` error just indicates that the image wasn't fully written to the SD card - if you look at @brianmituka 's screenshot closely you can see that Etcher hadn't even got to 1% before the card was 'unplugged', so it's not surprising that nothing could be mounted ;)

@jviotti If you don't mind I'll re-open this for now - as I mentioned in my previous comment `/dev/mmcblk0` indicates that it's (probably) an internal SD-slot, and I'd expect those to be higher-quality than cheap USB SD-adaptors, so I'd like to dig into this a bit deeper...

timdonovanuk, 2017-08-20:
I have this problem with Ethcher using an SD Card reader in Windows 7. Card reader has always worked 100% with other tools such as Win32DiskImager and I use it to transfer photos all the time with no issues.

lurch, 2017-08-20:
@timdonovanuk Are you definitely using the latest version (1.1.2) of Etcher? What's the exact error you see? Can you cut'n'paste your DevTools console logs please? (Press `Ctrl` and `Alt` and `i` at the same time). A screenshot of your error-message might be useful too.
Are you using an internal card reader, or an external (USB) card reader? Can you provide more details about both the SD card you're using and the SD card reader you're using?
If you have the time, can you verify that Win32DiskImager is successfully able to write the exact same disk-image to the exact same SD-card using the exact same card-reader, that Etcher is unable to?

I realise that's a lot of questions, but the more info you can provide, the easier it'll be for us to track down the exact problem ;-)

lurch, 2017-08-20:
Also, if @brianmituka could try testing again with the latest version of Etcher, that'd be really helpful :)

closed 2017-12-04 by jviotti

r4mbo7, 2018-12-04:
Fails after 1% with 1.4.8 just dl...
[image]

neodingo, 2021-08-21:
I see this behavior currently whenever the source image is on a network share on Windows 10.

lurch, 2021-08-21:
> I see this behavior currently whenever the source image is on a network share on Windows 10.

Duplicate of e.g. #3419 ?

neodingo, 2021-08-21:
That's my best guess, thanks for the link - I was hoping someone would send me to the latest on this.  I'll try again tomorrow and see if I can't gather some logs for additional data as I've got a 100% reproducible case.

awsome306, 2023-06-15:
Perhaps this is pure coincidence, but I was able to resolve this issue on my Windows 10 machine by running balenaEtcher as an administrator.

03stevensmi, 2023-10-18:
> Perhaps this is pure coincidence, but I was able to resolve this issue on my Windows 10 machine by running balenaEtcher as an administrator.

Thank you,
Where as everyone else here is full of sh** and assumes the worst that my pc is damaged...
And has nothing to do with the software...
Despite everyone saying others work etc etc...?
Sending me into panic mode...

This actually helped! 
"Run-as-administrator" sould in all be the first response or advice anytime this exact question gets asked.
Then if it doesn't work,
Go down the line of what was suggested above.
(Assuming others didn't work that is) ;)

Anyone reading this...
Download the portable version,
Disable controled folders in ransomware protection (its in the windows security virus settings)
(just temp ofc... Make sure when your finished to re-enable it in case you forget)
Right click the program and run as administrator,
Don't use url to flash (that never worked for me)
Select image from folder...
See if it worked for you...
If not...
Then continue here for "help"
Cheers awsome306!