balena-io/etcher #1495: fork resource temporarily unavailable error message on Mac Sierra

alberttwong, 2017-06-07:
- **Etcher version:** 1.0
- **Operating system and architecture:**  Mac Sierra 10.12.5
- **Image flashed:**  2017-04-10-raspbian-jessie.zip

This shows up when I'm trying to write a new image. 

Command failed: "/var/folders/by/4pgnv4fx5pv1n4hjc7b_5plr0000gn/T/tmp-65283gaqPPJQwc5AD.sh"
/var/folders/by/4pgnv4fx5pv1n4hjc7b_5plr0000gn/T/tmp-65283gaqPPJQwc5AD.sh: fork: Resource temporarily unavailable
 (code 128)

Error: Command failed: "/var/folders/by/4pgnv4fx5pv1n4hjc7b_5plr0000gn/T/tmp-65283gaqPPJQwc5AD.sh"
/var/folders/by/4pgnv4fx5pv1n4hjc7b_5plr0000gn/T/tmp-65283gaqPPJQwc5AD.sh: fork: Resource temporarily unavailable
 (code 128)
    at ChildProcess.exithandler (child_process.js:223:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:501:12)

lurch, 2017-06-07:
Maybe related to #1137 ?
Or possibly the OSX equivalent of #1054 or #1108 ?

jviotti, 2017-06-08:
According to https://stackoverflow.com/a/12079684, this can happen when running out of file descriptors. @alberttwong could this be the case? Can you try the commands in that answer to inspect the number of open file descriptors to check if you're reaching the limit when this error is thrown?

alberttwong, 2017-06-09:
```
Alberts-MacBook-Pro:~ alwong$ sysctl fs.file-nr
sysctl: unknown oid 'fs.file-nr'
Alberts-MacBook-Pro:~ alwong$ ulimit -Hn
unlimited
```

jviotti, 2017-06-10:
So as far as I can see (assuming I'm reading the output of `ulimit` correctly), you have an limit of 8192 open files, and 709 user processes, and either could be the culprit of the issue you are running into here.

Can you try increasing the maximum allowed open files as described in https://blog.dekstroza.io/ulimit-shenanigans-on-osx-el-capitan/, and see if that makes any effect?

Of course, we can't just tell people to increase their file limits just to use Etcher. We're currently moving the drive scanning routines (which are based on shell scripts) to native code, so we don't have to spawn anything, which should be the real solution.

@alberttwong BTW, can you show us the opened process after running Etcher for a while? I'm thinking you could be facing https://github.com/resin-io/etcher/issues/1288, which can cause a lot of perl processes to accumulate after a while.

valentinancy, 2017-10-15:
restart the terminal?

lurch, 2018-02-26:
I wonder if https://github.com/resin-io-modules/drivelist/pull/238 might have fixed this?

jhermsmeier, 2018-02-26:
Closing; see https://github.com/resin-io-modules/drivelist/issues/214#issuecomment-368598016 for details

closed 2018-02-26 by jhermsmeier