balena-io/etcher #1496: Windows Select Image status

tomsaul, 2017-06-07:
V1.0.0
Win10 (not important)

When initially selecting a (large) image, there is a long pause before the drive selection option is activated - there should be some kind of feedback to indicate the file load is in progress (hourglass, progress bar, ...).

lurch, 2017-06-07:
Can you provide a link to the image you're using? How big is it? What format is it in?

After selecting an image, Etcher tries to work out how big the image is (which may require seeking to a specific offset in a compressed file to read the uncompressed size), to ensure that only drives _smaller_ than this size are available for selection. I wouldn't expect this file-seeking to really take a significant amount of time though :-/

tomsaul, 2017-06-07:
It is a .img file -in this case about 2GB size.  FWIW, its an Armbian OS image if you want to grab one.  I am pulling it from a (slow) network drive.  It was taking 30-45 seconds, during which the app appeared to be hung.

lurch, 2017-06-07:
Ahhh, it'll be the network drive that's the issue then. Presumably Etcher works 'okay' if you copy the image to your local hard-drive?

Seems like there's many many Armbian OS images https://www.armbian.com/download/ - could you link us to the exact one you're using please?

BTW, in terms of typical Etcher usage, I'd say 2GB is on the small(er) size, rather than the 'large' size ;)

I agree that it probably makes sense for Etcher to display an hourglass cursor in the situation you're describing though.

tomsaul, 2017-06-08:
They should all be similar, but as I recall it was https://dl.armbian.com/orangepiplus2e/Ubuntu_xenial_default_desktop.7z

In any case it is working, it is just confusing when the app becomes unresponsive during the load time.

lurch, 2017-06-08:
Ahhh, a shame that Armbian is still using `.7z` files (see #711 ).

I guess that means that you have to uncompress the file first (making it take much longer to load over the network), and unlike with `.xz` you can't just load the compressed file straight into Etcher.

closed 2018-10-15 by jviotti