balena-io/etcher #1507: can't recycle SD previously flashed with OSMC

RogueM, 2017-06-08:
- **Etcher version: 1.0.0 (repo sync tonight)**
- **Operating system and architecture: Debian Stretch (armv7) running in chroot on ChromeBook**
- **Image flashed: OSMC**
- **Do you see any meaningful error information in the DevTools?**
(don't know how to get up on this platform)

when reusing a SD previously flashed with OSMC, etcher fails early in process with following error:

Error: Child process exited with error code: 1
    at ChildProcess.child.on (/home/rogue/etcher/lib/child-writer/index.js:217:24)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)

lurch, 2017-06-08:
If you can reproduce this on a 'standard' Linux PC (or Mac or Windows) using one of the builds downloaded from https://etcher.io that'd be really helpful. And which version of OSMC (a link to a particular download would help) had previously been on the SD card?

Do the instructions from https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives help you get over this particular hiccup?
Although if you find this error to be reproducible that would be _really_ useful :)

@jviotti I guess this issue counts as another 'vote' for #1362 :wink:

RogueM, 2017-06-08:
I'll check into it Andrew... in the meantime, here's the output from the commands suggested in the other (unrelated, but still) topic:

(stretch)rogue@localhost:~$ lsblk -b -d "/dev/mmcblk1" --output SIZE,RO,RM,MODEL 
      SIZE RO RM MODEL
3965190144  0  0 

(stretch)rogue@localhost:~$ udevadm info --query=property --export --export-prefix=UDEV_ --name="/dev/mmcblk1"
UDEV_DEVLINKS='/dev/disk/by-path/platform-ff0c0000.dwmmc /dev/disk/by-id/mmc-SU04G_0x005c8746'
UDEV_DEVNAME='/dev/mmcblk1'
UDEV_DEVPATH='/devices/ff0c0000.dwmmc/mmc_host/mmc1/mmc1:aaaa/block/mmcblk1'
UDEV_DEVTYPE='disk'
UDEV_ID_NAME='SU04G'
UDEV_ID_PART_TABLE_TYPE='dos'
UDEV_ID_PART_TABLE_UUID='f519a7af'
UDEV_ID_PATH='platform-ff0c0000.dwmmc'
UDEV_ID_PATH_TAG='platform-ff0c0000_dwmmc'
UDEV_ID_SERIAL='0x005c8746'
UDEV_MAJOR='179'
UDEV_MINOR='0'
UDEV_NPARTS='2'
UDEV_SUBSYSTEM='block'
UDEV_USEC_INITIALIZED='8192731'

(stretch)rogue@localhost:~$ mount (edited)
/dev/mmcblk1p1 on /var/host/media/removable/SD Card type vfat (rw,nosuid,nodev,noexec,relatime,dirsync,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,flush,errors=remount-ro)
/dev/mmcblk1p2 on /var/host/media/removable/SD Card 1 type ext4 (rw,nosuid,nodev,noexec,relatime,dirsync,seclabel,stripe=1024,data=ordered)

RogueM, 2017-06-09:
hum, I picked up another SD with prior Raspbia and same problem. So right now I can't use Etcher at all on this machine. So I don't think the issue is related to the partitions and filesystem on the target.

I'll try again tomorrow to get something successfully flashed tomorrow, as it stands it seems that what's in the tip of the master branch is not playing ball here.

lurch, 2017-06-09:
Interesting that you've got the SD card appearing as `/dev/mmcblk1` - it's the first time I've seen `/dev/mmcblk1` rather than `/dev/mmcblk0`.

I wonder if the problem here might be something to do with chroot?
I'll leave this issue open for now, but we might end up closing it as invalid if it's specific to "chroot on ChromeBook".

RogueM, 2017-06-09:
the chromebook boots from flash, `/dev/mmcblk0` is the internal storage. I had etcher working from version 0.0.8 to now, including initial state of sources for 1.0.0 - no doubt the chroot makes things more complex though (I have to `sudo ls` prior to launching etcher, to get write access, for a start) but it worked until now. Of course it could be the underlying OS that has topped it over the edge.

jviotti, 2017-06-10:
@RogueM Thanks for all the information so far. Its still tricky to provide any meaningful without playing around with a ChromeBook myself though.

Some ideas:

- Can you try completely erasing the file-systems of the SD Card (thus leaving the SD Card in a blank state), and checking if Etcher runs fine after that?

> I'll try again tomorrow to get something successfully flashed tomorrow, as it stands it seems that what's in the tip of the master branch is not playing ball here.

So you're building from the tip of `master`? Can you try building from older tags (e.g. `v1.0.0`, `v1.0.0-beta.19`, etc)? If we can identify the first version that introduced the issue (in case this was not happening before), then that should make wonders to identify the "bad commit."

closed 2017-12-04 by jviotti