balena-io/etcher #1519: Symantec Endpoint Protection prevents Etcher from running

masieran, 2017-06-14:
- **Etcher version:** **Etcher-1.0.0-win32-x64.zip**
- **Operating system and architecture:** Windows 7 Pro 64, i5-4310U
- **Image flashed:** none
- **Do you see any meaningful error information in the DevTools?** no

I have unzipped Etcher-1.0.0-win32-x64.zip to an empty directory on Windows 7 Pro 64. Then I clicked on Etcher.exe. The program execution was terminated immediately by Symantec Endpoint Protection that is commonly used in the corporate environment. The Symantec program quarantined the lzma_native.node file. 

Risk: WS.Reputation.1
Filename: lzma_native.node
Type: Quarantine
Original location: c:\Users\...
Status: Infected
Date: 6/13/2017 5:13 PM
File Create Date: 5/12/2017 2:43 AM

masieran, 2017-06-14:
[image]

masieran, 2017-06-16:
There is a work around  Etcher-1.0.0-win32-x64.exe has different version of lzma_native.node that doesn't trigger Symantec Endpoint Protection.
$ md5sum lzma*
8de14ce3c640cfc36983a746f7678c41 *lzma_native.node  - good
e1600afc820baf70a899e0f4e269dcc9 *lzma_native.node1 - bad

jhermsmeier, 2017-06-16:
Hmm, this seems odd – @jviotti any idea how `lzma_native.node` could hash to a different value from a .zip, than from the installer?

Regarding the quarantine – **WS.Reputation.1**: *"files that have a low reputation score based on analyzing data from Symantec’s community of users and therefore are likely to be security risks."* (Details: https://www.symantec.com/security_response/writeup.jsp?docid=2010-051308-1854-99)

Which basically means that Symantec Endpoint Protection will label everything as "infected" it's never seen before. This is effing ridiculous. If they'd at least label it as "unknown binary" or something, instead of fearmongering.

@masieran Looks like you're one of the first to run it in that environment :) But sadly this also seems to mean, we can't do much about it, other than wait until their splendid "reputation-based security technology" has seen enough of Etcher to stop labelling parts of it as infected.

masieran, 2017-06-16:
The exe and zip variants of lzma_native.node have the same size, but they are different::
$ diff exe.hex zip.hex
18c18
< 000110 50 45 00 00 64 86 07 00 **58 5a** 15 59 00 00 00 00
---
> 000110 50 45 00 00 64 86 07 00 **6a 58** 15 59 00 00 00 00
9628c9628
< 025c80 00 00 00 00 **58 5a** 15 59 00 00 00 00 02 00 00 00
---
> 025c80 00 00 00 00 **6a 58** 15 59 00 00 00 00 02 00 00 00
9630,9631c9630,9631
< 025ca0 **58 5a** 15 59 00 00 00 00 0c 00 00 00 14 00 00 00
< 025cb0 00 71 02 00 00 61 02 00 00 00 00 00 **58 5a** 15 59
---
> 025ca0 **6a 58** 15 59 00 00 00 00 0c 00 00 00 14 00 00 00
> 025cb0 00 71 02 00 00 61 02 00 00 00 00 00 **6a 58** 15 59
9689,9690c9689,9690
< 026070 00 00 00 00 52 53 44 53 **62 58 b7 53 85 2b f8 45**
< 026080 **b0 aa 4e 7f dc 71 2c bf** 01 00 00 00 63 3a 5c 50
---
> 026070 00 00 00 00 52 53 44 53 **25 93 62 30 7e 13 6e 4b**
> 026080 **8f 6c 68 87 f2 f6 32 a1** 01 00 00 00 63 3a 5c 50
10206c10206
< 0280c0 00 00 00 00 **57 5a** 15 59 00 00 00 00 f2 90 02 00
---
> 0280c0 00 00 00 00 **69 58** 15 59 00 00 00 00 f2 90 02 00
@jhermsmeier, I would restate your question: "What makes the exe variant of lzma_native.node to be considered as "reputable" by the Symantec Endpoint Protection? :-)

jhermsmeier, 2017-06-16:
@masieran our current theory is that the installer & the portable version have been built on two different machines with different Windows & possibly SDK versions, hence the difference. Haven't verified that yet, though.

jhermsmeier, 2017-06-16:
I've now filed a whitelisting request for this particular case (Etcher Win x64 Portable) with Symantech – waiting for a response now (they say it'll take ~2 days to review).

jviotti, 2017-06-16:
I'll see if we can code-sign the native add-ons as well, which should probably help reduce these kinds of incidents.

lurch, 2017-06-17:
> I'll see if we can code-sign the native add-ons as well

Will that still be necessary after #1518 ? Or will it only be the "outer-most" executable that needs to be signed?

jviotti, 2017-06-21:
Yeah, I guess in the case of the portable version we only need to sign the top level executable.

jviotti, 2017-06-21:
So it seems that we can sign `*.node` files as well, and node loads them just fine, so I'll rename this issue to reflect the task of signing add-ons.

jhermsmeier, 2017-06-22:
@masieran btw, the whitelisting request came through, and they've added the binary in question – so this particular release shouldn't cause trouble anymore.

closed 2018-10-15 by jviotti