balena-io/etcher #1522: Doesn't detect USB drive in macOS sierra

iattinder, 2017-06-16:
version-1.0.0
macOS- 10.12.5 (16F73)
image- Samsung 850 pro firmware.iso
USB device- Sandisk Extreme USB 16 GB

It doesn't even detect the drive let alone flashing it.

jviotti, 2017-06-16:
Hi @iattinder ,

Did you try enabling unsafe mode in the settings page? If that worked, can you show me the output of `diskutil info /dev/diskN`, where `N` is the disk number of the problematic drive?

iattinder, 2017-06-24:
No,

Attinder-MacBook-Pro:~ Attinder$ diskutil info /dev/disk3
   Device Identifier:        disk3
   Device Node:              /dev/disk3
   Whole:                    Yes
   Part of Whole:            disk3
   Device / Media Name:      Extreme

   Volume Name:              Not applicable (no file system)
   Mounted:                  Not applicable (no file system)
   File System:              None

   Content (IOContent):      FDisk_partition_scheme
   OS Can Be Installed:      No
   Media Type:               Generic
   Protocol:                 USB
   SMART Status:             Not Supported

   Disk Size:                15.7 GB (15693664256 Bytes) (exactly 30651688 512-Byte-Units)
   Device Block Size:        512 Bytes

   Read-Only Media:          No
   Read-Only Volume:         Not applicable (no file system)

   Device Location:          External
   Removable Media:          Fixed

   Virtual:                  No
   OS 9 Drivers:             No
   Low Level Format:         Not supported

jviotti, 2017-06-26:
Hi @iattinder ,

Thanks for the information! Looks like macOS itself is detecting your drive as non-removable:

```
   Removable Media:          Fixed
```

Sadly there's not much we can do if that's what the OS reports, so I advise you to research how to make macOS correctly detect that device as a removable drive, and then Etcher will detect it correctly.

In the worst case, looks like enabling "Unsafe Mode" (in the settings page from the cog at the top right corner) should help you out.

closed 2017-06-26 by jviotti

iattinder, 2017-06-26:
Well Mac Linux USB loader does detect it. Unsafe Mode doesn't work. UNetbootin doesn't work too.

jviotti, 2017-06-26:
Can you try running this script: https://github.com/resin-io-modules/drivelist/blob/master/scripts/darwin.sh while the problematic drive is plugged in, and paste the output here?

iattinder, 2017-06-28:
device: /dev/disk0
description: "Samsung SSD 850 PRO 256GB"
size: 256060514304
mountpoints: []
raw: /dev/rdisk0
protected: False
system: True

device: /dev/disk1
description: "Samsung SSD - Samsung SSD 850 PRO 256GB"
size: 235479527424
mountpoints:
  - path: "/"
raw: /dev/rdisk1
protected: False
system: True

device: /dev/disk2
description: "TOSHIBA MK7559GSXF"
size: 750156374016
mountpoints:
  - path: "/Volumes/Toshiba HDD"
raw: /dev/rdisk2
protected: False
system: True

device: /dev/disk3
description: "Extreme"
size: 15693664256
mountpoints:
  - path: "/Volumes/Sandisk"
raw: /dev/rdisk3
protected: False
system: True

logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

jviotti, 2017-06-28:
So the drive is `/dev/disk3`, which is detected by the drive scanning routine, but as a system drive. 

I'll be closing this issue. Unsafe mode should work fine in this case, and there's not much we can do to detect it as a removable drive if macOS itself fails to detect it as such.

lurch, 2017-06-28:
@iattinder Could you clarify your "Unsafe Mode doesn't work" comment? When you enable Etcher's Unsafe Mode in the settings, which drives does Etcher then display in the drive-selection dialog? (A screenshot would be great)