balena-io/etcher #1523: Validate write on success reads data from cache instead of SD card

net147, 2017-06-16:
- **Etcher version:**
1.0.0
- **Operating system and architecture:**
Linux x64
- **Image flashed:**
150995968 byte image file
- **Do you see any meaningful error information in the DevTools?**
It does not show any "Progress (check)" messages in Console like it does on Windows

Validate write finishes immediately without actually reading from the SD card.

net147, 2017-06-16:
If I run the following command in another terminal to keep flushing all the caches, I can Etcher reading from the SD card:
while true; do echo 3 > /proc/sys/vm/drop_caches; done

lurch, 2017-06-19:
If true, this should be regarded as a high-priority bug, as I guess it means validation isn't doing what we actually think it is :-/
Perhaps there's some extra options we can pass to the 'open' call to get it to avoid cached reads?

lurch, 2017-06-19:
Hmmm, just done some searching and found https://stackoverflow.com/questions/15266115/read-file-without-disk-caching-in-linux/15266168#15266168 and also found https://nodejs.org/api/fs.html#fs_fs_open_path_flags_mode_callback which says
```
'rs+' - Open file for reading and writing in synchronous mode. Instructs the operating system to bypass the local file system cache.
```
And `rs+` is indeed what we're using: https://github.com/resin-io/etcher/blob/master/lib/cli/writer.js#L76

:confused:

lurch, 2017-06-19:
...although https://github.com/nodejs/node/blob/master/test/parallel/test-fs-open-flags.js#L42 implies that `rs+` only sets `O_RDWR | O_SYNC` and we'll instead need to pass the `flags` to `open()` as a number instead of a string, if we want to use `O_DIRECT` https://nodejs.org/api/fs.html#fs_file_open_constants
And see also https://linux.die.net/man/2/open

But it looks like `O_DIRECT` is specific to Linux, and the equivalent on Windows is `FILE_FLAG_NO_BUFFERING` https://msdn.microsoft.com/en-us/library/windows/desktop/cc644950(v=vs.85).aspx
And these links are useful too https://stackoverflow.com/questions/924900/portability-of-open-o-direct-in-c and https://stackoverflow.com/questions/2299402/how-does-one-do-raw-io-on-mac-os-x-ie-equivalent-to-linuxs-o-direct-flag

Looks like this gets quite complex, low-level and platform-specific very quickly :-/  Perhaps the 'best' solution would be some kind of UnbufferedFileReaderWriter native C++ nodejs addon? Which I _think_ is sort of the direction we were planning on heading in (so that we can do 'proper' locking on Windows) anyway...? Unless @petrosagg has any clever suggestions?

net147, 2017-06-19:
It seems like either the Node.js documentation needs to be fixed or Node.js needs to be fixed to reflect the documentation.

jviotti, 2017-06-19:
Hi @net147,

Can you try the following custom Etcher build: https://misc1.dev.resin.io/~jviotti/direct-flag/Etcher-1.0.0+dbc8aa1-linux-x64.zip? It includes the following patch:

```diff
diff --git a/lib/cli/writer.js b/lib/cli/writer.js
index f8a966b..a46c212 100644
--- a/lib/cli/writer.js
+++ b/lib/cli/writer.js
@@ -73,7 +73,7 @@ exports.writeImage = (imagePath, drive, options, onProgress) => {

     return mountutils.unmountDiskAsync(drive.device);
   }).then(() => {
-    return fs.openAsync(drive.raw, 'rs+');
+    return fs.openAsync(drive.raw, fs.constants.O_RDWR | fs.constants.O_SYNC | fs.constants.O_DIRECT);
   }).then((driveFileDescriptor) => {
     return imageStream.getFromFilePath(imagePath).then((image) => {
       if (!constraints.isDriveLargeEnough(drive, image)) {
```

If this works for you, then we can add the `O_DIRECT` flag only for Linux (plus work with the Node.js team to get their docs fixed).

net147, 2017-06-20:
@jviotti I get an error dialog with the message "EINVAL: invalid argument, write" after it finishes writing.

net147, 2017-06-20:
I suspect the buffers need to be aligned for O_DIRECT (e.g. https://github.com/bobrik/node-aligned-buffer)

lurch, 2017-06-20:
> I suspect the buffers need to be aligned for O_DIRECT

Yeah, it's not as simple as 'just' adding that flag @jviotti , which is why I said that this gets quite complex very quickly ;-)

> https://github.com/bobrik/node-aligned-buffer

Oooh, that looks interesting, but unfortunately it seems that it's only for reading and doesn't offer writing.
**Although**, maybe using `O_SYNC` on writing, and `O_DIRECT` on reading, _is_ in fact all we need to get around the problem being described here?
https://stackoverflow.com/questions/5055859/how-are-the-o-sync-and-o-direct-flags-in-open2-different-alike

Although I guess `node-aligned-buffer` probably still only works on Linux, and not OSX or Windows (and it was last updated 5 years ago!). And strictly speaking we _should_ (could) use `O_DIRECT` (and equivalents) for writing too, as there's no point filling up the filesystem cache with the disk-image that Etcher is writing. And I guess using `node-aligned-buffer` would also mean that we'd need to close and re-open the file-handle between writing and reading.

> we can add the O_DIRECT flag only for Linux

If the image being flashed is small enough to fit in the filesystem cache (the image @net147 is flashing here is only 144 MB), then I see no reason why Windows or OSX wouldn't suffer from this bug in exactly the same way.

lurch, 2017-06-20:
Using a file created with `dd if=/dev/urandom of=~/Downloads/small_image.img count=294914` and a fairly old (and therefore slow) 2GB USB flash drive, I've now been able to replicate this issue myself.

DevTools log on Linux:
```
Tue Jun 20 2017 14:21:52 GMT+0100 (BST) Progress (write): 5% at 7.8 MB/s (eta 18s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:53 GMT+0100 (BST) Progress (write): 7% at 8 MB/s (eta 17s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:53 GMT+0100 (BST) Progress (write): 10% at 7.19 MB/s (eta 19s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:54 GMT+0100 (BST) Progress (write): 13% at 7.41 MB/s (eta 18s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:54 GMT+0100 (BST) Progress (write): 16% at 7.56 MB/s (eta 17s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:55 GMT+0100 (BST) Progress (write): 19% at 7.67 MB/s (eta 16s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:56 GMT+0100 (BST) Progress (write): 21% at 7.33 MB/s (eta 16s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:56 GMT+0100 (BST) Progress (write): 24% at 7.43 MB/s (eta 15s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:57 GMT+0100 (BST) Progress (write): 27% at 7.19 MB/s (eta 15s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:57 GMT+0100 (BST) Progress (write): 28% at 6.96 MB/s (eta 15s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:58 GMT+0100 (BST) Progress (write): 31% at 7.07 MB/s (eta 15s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:59 GMT+0100 (BST) Progress (write): 34% at 6.92 MB/s (eta 14s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:21:59 GMT+0100 (BST) Progress (write): 37% at 7.01 MB/s (eta 14s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:00 GMT+0100 (BST) Progress (write): 39% at 6.89 MB/s (eta 13s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:01 GMT+0100 (BST) Progress (write): 42% at 6.79 MB/s (eta 13s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:01 GMT+0100 (BST) Progress (write): 45% at 6.87 MB/s (eta 12s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:02 GMT+0100 (BST) Progress (write): 48% at 6.78 MB/s (eta 12s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:03 GMT+0100 (BST) Progress (write): 51% at 6.85 MB/s (eta 11s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:03 GMT+0100 (BST) Progress (write): 53% at 6.91 MB/s (eta 10s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:04 GMT+0100 (BST) Progress (write): 55% at 6.84 MB/s (eta 10s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:04 GMT+0100 (BST) Progress (write): 57% at 6.6 MB/s (eta 10s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:05 GMT+0100 (BST) Progress (write): 59% at 6.67 MB/s (eta 9s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:05 GMT+0100 (BST) Progress (write): 62% at 6.59 MB/s (eta 9s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:06 GMT+0100 (BST) Progress (write): 64% at 6.54 MB/s (eta 8s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:07 GMT+0100 (BST) Progress (write): 67% at 6.6 MB/s (eta 7s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:07 GMT+0100 (BST) Progress (write): 70% at 6.65 MB/s (eta 7s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:08 GMT+0100 (BST) Progress (write): 72% at 6.6 MB/s (eta 6s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:08 GMT+0100 (BST) Progress (write): 74% at 6.63 MB/s (eta 6s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:09 GMT+0100 (BST) Progress (write): 77% at 6.68 MB/s (eta 5s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:09 GMT+0100 (BST) Progress (write): 79% at 6.63 MB/s (eta 5s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:22:10 GMT+0100 (BST) Progress (write): 81% at 6.57 MB/s (eta 4s)
[... cut ...]
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:23:36 GMT+0100 (BST) Progress (check): 77% at 13.71 MB/s (eta 3s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:23:37 GMT+0100 (BST) Progress (check): 82% at 13.76 MB/s (eta 2s)
/usr/lib/etcher-electron/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Tue Jun 20 2017 14:23:37 GMT+0100 (BST) Progress (check): 86% at 13.82 MB/s (eta 1s)
```
(more in line with what I'd expect for this old USB drive: multiple `check` lines, reading at around 13 MB/s).

lurch, 2017-06-20:
...and here's the DevTools log from a (less powerful) Windows 8 computer, writing exactly the same image to exactly the same USB flash drive:
```
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:54 GMT+0100 (GMT Daylight Time) Progress (write): 3% at 5.77 MB/s (eta 25s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:55 GMT+0100 (GMT Daylight Time) Progress (write): 5% at 5.77 MB/s (eta 25s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:55 GMT+0100 (GMT Daylight Time) Progress (write): 9% at 6.29 MB/s (eta 22s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:56 GMT+0100 (GMT Daylight Time) Progress (write): 12% at 6.67 MB/s (eta 20s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:56 GMT+0100 (GMT Daylight Time) Progress (write): 14% at 6.94 MB/s (eta 19s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:57 GMT+0100 (GMT Daylight Time) Progress (write): 18% at 6.93 MB/s (eta 18s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:58 GMT+0100 (GMT Daylight Time) Progress (write): 19% at 6.64 MB/s (eta 18s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:58 GMT+0100 (GMT Daylight Time) Progress (write): 23% at 7.01 MB/s (eta 17s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:59 GMT+0100 (GMT Daylight Time) Progress (write): 25% at 7.03 MB/s (eta 16s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:48:59 GMT+0100 (GMT Daylight Time) Progress (write): 28% at 7.12 MB/s (eta 15s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:00 GMT+0100 (GMT Daylight Time) Progress (write): 30% at 7.17 MB/s (eta 15s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:00 GMT+0100 (GMT Daylight Time) Progress (write): 32% at 7.12 MB/s (eta 14s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:01 GMT+0100 (GMT Daylight Time) Progress (write): 35% at 7.2 MB/s (eta 13s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:01 GMT+0100 (GMT Daylight Time) Progress (write): 38% at 7.05 MB/s (eta 13s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:02 GMT+0100 (GMT Daylight Time) Progress (write): 41% at 7.12 MB/s (eta 12s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:02 GMT+0100 (GMT Daylight Time) Progress (write): 44% at 7.19 MB/s (eta 12s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:03 GMT+0100 (GMT Daylight Time) Progress (write): 46% at 7.25 MB/s (eta 11s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:04 GMT+0100 (GMT Daylight Time) Progress (write): 49% at 7.31 MB/s (eta 10s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:04 GMT+0100 (GMT Daylight Time) Progress (write): 51% at 7.17 MB/s (eta 10s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:05 GMT+0100 (GMT Daylight Time) Progress (write): 53% at 7.22 MB/s (eta 10s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:05 GMT+0100 (GMT Daylight Time) Progress (write): 56% at 7.12 MB/s (eta 9s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:06 GMT+0100 (GMT Daylight Time) Progress (write): 59% at 7.17 MB/s (eta 9s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:06 GMT+0100 (GMT Daylight Time) Progress (write): 62% at 7.21 MB/s (eta 8s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:07 GMT+0100 (GMT Daylight Time) Progress (write): 64% at 7.26 MB/s (eta 7s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:07 GMT+0100 (GMT Daylight Time) Progress (write): 67% at 7.24 MB/s (eta 7s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:08 GMT+0100 (GMT Daylight Time) Progress (write): 70% at 7.34 MB/s (eta 6s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:08 GMT+0100 (GMT Daylight Time) Progress (write): 71% at 7.24 MB/s (eta 6s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:09 GMT+0100 (GMT Daylight Time) Progress (write): 74% at 7.27 MB/s (eta 5s)
[... cut ...]
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:33 GMT+0100 (GMT Daylight Time) Progress (check): 92% at 8.01 MB/s (eta 1s)
C:\Program Files (x86)\Etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:51 Tue Jun 20 2017 15:49:33 GMT+0100 (GMT Daylight Time) Progress (check): 95% at 8.01 MB/s (eta 1s)
```
I guess the buffering must work differently in Windows to the way it does in Linux :-)
Unfortunately I don't have a Mac with which I can test the OSX behaviour.

lurch, 2017-09-29:
#751 mentioned https://github.com/ronomon/direct-io , which looks interesting :)

Teesla, 2017-10-20:
I can confirm this bug still present in v1.1.2

lurch, 2017-11-19:
@jhermsmeier I guess this is something that we probably ought to fix in time for EtcherPro?
I guess at the very least we could run `sync; echo 3 > /proc/sys/vm/drop_caches` on Linux (the latter part needs root permissions) between the end of the writing phase and the start of the verifying phase?

petrosagg, 2017-11-19:
@lurch no this is a very bad idea because it will invalidate every single page in the cache which will cause performance issues for all the other applications running on the system. Use direct IO or maybe experiment with `posix_fadvise` to only bypass the cache for the particular device

lurch, 2017-11-19:
Yeah, I agree that direct IO is the "proper" way to achieve this.

lurch, 2017-12-07:
Fixed by #1875 (which will be included in Etcher v1.2.1) but please leave additional comments if you're still seeing this behaviour in Etcher v1.2.1

lurch, 2018-02-23:
Unfortunately #2055 reintroduces this issue.

jhermsmeier, 2018-03-01:
Note to self: We can potentially work around this particularity by opening a new handle with different flags for verification.

lurch, 2018-03-01:
I thought about that once, but doesn't closing the write-handle and then opening a read-handle mean that there's a (very small) time-interval where the drive-handle is closed, at which point the OS might jump in and auto-mount the partitions? \*shrug\*

jhermsmeier, 2018-03-01:
@lurch yeah, that might be; but at that point it shouldn't interfere with anything following that – could also open the new handle while the previous one isn't closed yet. Definitely needs some fiddeling & testing.

lurch, 2018-03-02:
> but at that point it shouldn't interfere with anything following that

...except that some FSes store a "last mount time" in their metadata, which means that if that partition gets auto-mounted the on-disk data is then different from the original image, causing the verification to fail ;-) https://github.com/resin-io/etcher/issues/2085#issuecomment-368819762

> Definitely needs some fiddeling & testing.

I'll keep my fingers crossed for you! :-D

closed 2018-10-15 by jviotti

lurch, 2019-06-12:
Not sure why this was closed, it's definitely still an issue. I just flashed a 5GB image onto a (slightly older) 8GB microSD card, using Etcher 1.5.47 on my Ubuntu 16.04 desktop PC with an i7 CPU and 32 GB of RAM. The image wrote at a fairly constant 7MB/s, but apparently verified at over 500MB/s ! :open_mouth: