balena-io/etcher #1525: Error: unable to get local issuer certificate

kodweis, 2017-06-16:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** Linux ghost 4.8.0-54-generic #57-Ubuntu SMP Wed May 24 10:21:44 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** : [screenshots](http://imgur.com/a/t0Co7)

It still writes images ok.. just this message pops up every time I open the app. Probably has to do with corporate proxy I'm in

jviotti, 2017-06-16:
Hi @kodweis ,

Are you behind a proxy or something like that? As the error message states, you should update your system root SSL certificates (`ca-certificates` package).

This can also happen because of an un-trusted certificate, but I'm ruling out that possibility since its unlikely Amazon S3 will have such a certificate.

jviotti, 2017-06-16:
> Probably has to do with corporate proxy I'm in

Oh, I missed that part. Yeah, looks like the proxy might not be well configured and might be messing up with the connection's SSL certificates.

jviotti, 2017-08-01:
There's not much we can do from our side here. I only see two options:

- Swallow the error (and maybe log it in DevTools just for debugging purposes)
- Present a friendly warning window describing what happened, every time Etcher starts

What do you think?

kodweis, 2017-08-02:
What functionality is lost due to this error? An update check?
Since it's not a critical error that prevents the app from performing it's main purpose, I would say just log it to DevTools. I don't think this error warrants a popup dialog every time someone starts the app.

jviotti, 2017-08-02:
Yeah, its just an update check. I agree with you, and will make the change soon.

jviotti, 2017-08-02:
Hi @kodweis , I agree. Thanks a lot for the feedback. I implemented the discussed changes, which should land in the next version.

kodweis, 2017-08-02:
Awesome! Thanks!

closed 2017-08-03 by jviotti