balena-io/etcher #1526: On OSX, upon starting flash, I get a dialog that "Etcher want to make changes"?

dteeter, 2017-06-16:
Why is this giving this type of permission necessary? What is Etcher doing in the back ground if I give that permission?
[image]

deadair3210, 2017-06-17:
If OS X is anything like Linux and Windows permission wise, its probably requesting permissions to alter the drive at low level. Needed for block level operations

jviotti, 2017-06-17:
Hi @dteeter ,

Writing to raw external drives require superuser permissions in all unix based operating systems, including macOS. Etcher is open source, so you can take a look if you want to learn exactly what's happening under the hood.

closed 2017-06-17 by jviotti