balena-io/etcher #1529: "No polkit authentication agent found" ERROR

5pi3ltaunt, 2017-06-18:
I am running Etcher x64 version 1.0.0 on a Acer Chromebook running Ubuntu Xfce4.
I attempted to flash a SD card with Pwnpi 3.0 for the Raspberry pi 3.
The DevTools are not coming up and i don't know why, I am incredibly sorry for the lack of information.
I successfully downloaded the Ubuntu Xfce4 polkits (I say polkits because i downloaded about 5 different ones) the ERROR is still coming up. 
I have read Issue 1179, but the information given, I assume, is for if you have downloaded it using a "git clone" command, which i did not do. I downloaded it at the website and i don't know if that has any different properties. I am new to github as well so please don't criticize my lack of code references. : ) hope you can help

lurch, 2017-06-19:
Have you read https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#no-polkit-authentication-agent-found-error-in-gnulinux ?

jviotti, 2017-06-19:
Hi @5pi3ltaunt ,

Can you show me that are the exact polkit packages that you have installed in your system? Any of these should suffice: polkit-1-auth-agent, policykit-1-gnome, polkit-kde-1, mate-polkit-bin.

Try installing all the ones mentioned above, and if that doesn't work, I'd check your system's polkit configuration, and see if there is something that could be causing polkit to not find/detect the available agents.

5pi3ltaunt, 2017-06-19:
I have policykit-1-gnome, polkit-kde-1, and mate-polkit-bin of the ones mentioned, I also have an xfce polkit from the ubuntu website installed.and if you could tell me how to access my system's polkit config that would be great, because i dont want to search on the internet more than i already have.

jviotti, 2017-06-21:
I installed Xubuntu here, and Etcher runs fine on a fresh installation:

[image]

There are the PoliciKit related packages I have on my system:

```
libpolkit-agent-1-0/xenial,now 0.105-14.1 amd64 [installed]
libpolkit-backend-1-0/xenial,now 0.105-14.1 amd64 [installed]
libpolkit-gobject-1-0/xenial,now 0.105-14.1 amd64 [installed]
policykit-1/xenial,now 0.105-14.1 amd64 [installed]
policykit-1-gnome/xenial,now 0.105-2ubuntu2 amd64 [installed]
policykit-desktop-privileges/xenial,xenial,now 0.20 all [installed]
```

> because i dont want to search on the internet more than i already have, because i dont want to search on the internet more than i already have.

Configuration files are under `/etc`, and checking the `polkit(8)` man page should teach you everything you need to configure polkit in your system.

I'm closing this since there is definitely a polkit configuration issue in your system, and there's no way we can help any further. Etcher simply spawns `pkexec`, so if you manage to get that working, then Etcher will succeed.

In the worst case, you can execute the AppImage as `sudo`, and thus make the elevation system unnecessary.

closed 2017-06-21 by jviotti