balena-io/etcher #1532: Integrate pipage in etcher-image-write

jviotti, 2017-06-20:
See https://github.com/resin-io-modules/pipage

lurch, 2017-06-20:
Should there be a 'linked' issue at https://github.com/resin-io-modules/etcher-image-write/issues ?

jviotti, 2017-06-21:
Up to @jhermsmeier. IMHO, is nicer to unify tickets in the main Etcher project, since its easier to search/tag everything on a single place, but I have no strong opinions neither.

jhermsmeier, 2017-06-21:
@lurch the pipage stuff is happening in Etcher, and `etcher-image-write` won't be used anymore at that point, hence it'd make sense to only track it here.

lurch, 2017-06-21:
Ah, so the title should be changed from `Integrate pipage in etcher-image-write` to `Integrate pipage in Etcher` ? ;-) Or even `Replace etcher-image-write with pipage` ?

jviotti, 2017-06-22:
I rephrased etcher-image-write to "the writer."

jhermsmeier, 2017-11-17:
Done in #1671

closed 2017-11-17 by jhermsmeier