balena-io/etcher #1545: Etcher not using Kdialog and working properly in sudo

now-im, 2017-06-24:
- **Etcher version: 1.0.0_1-1**
- **Operating system and architecture: 4.9.33-1-MANJARO(KDE)**
- **Image flashed: Qubes-R3.2-x86_64.iso**

Etcher is working properly in direct **"etcher"** input in command line. But there are the two major issues I am facing when I try to run **"sudo etcher"**. I wanted to try this because etcher needs root permission anyway.

But, then when I try to select an image instead of kdialog, gdialog pops up.
[image]
then, image flashing gets stuck at some point. From terminal I am seeing some repeating errors. They are:
```
(etcher:29851): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(etcher:29851): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(etcher:29851): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(etcher:29851): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(etcher:29851): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(etcher:29851): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(etcher:29851): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
```
Etcher has become a worthy replacement of rufus in linux for me. Hope this can help it someway.

lurch, 2017-06-24:
What does "Etcher version: 1.0.0_1-1" mean? Where / how did you download / install it?

now-im, 2017-06-24:
I installed it from arch user repository. You can see how the version has changed here:
https://aur.archlinux.org/cgit/aur.git/log/?h=etcher

jviotti, 2017-06-24:
Hi @now-im ,

As far as I know, the Electron framework is built on GTK (I don't think it supports Kdialog). Are you saying that when you run Etcher as your user, Kdialog is used, but then you run it with `sudo`, the GTK version pops up?

now-im, 2017-06-25:
@jviotti Sorry, I have made a mistake. When running normally gtk3 based gdialog is used and when running in sudo gtk2 based gdialog is being run and that is causing the problem. Normally, etcher doesn't leave any command output and gdialog and kdialog under same theme looks similar. So I thought, it was using kdialog. My bad but in sudo the problem still exists.

lurch, 2017-06-26:
> in sudo the problem still exists.

I assume you're referring to the brief mention of `image flashing gets stuck at some point` in your first post? Could you provide some more details please? Does the image flashing _always_ work when you run just `etcher` (without sudo, when it self-elevates), and does it _always_ fail when you run `sudo etcher` ? What messages do you see in the DevTools console? (press `Ctrl` + `Alt` + `i` at the same time)

Does it always fail in the same place? Do you see the same behaviour with different SD cards?

jviotti, 2017-06-26:
@now-im Electron/Chromium are built on GTK+2 only. Could it be that your user has a GTK configuration that includes a theme + other settings to make GTK+2 applications look like GTK+3? Then, when you run as `sudo`, that user specific configuration is not read, so then you see it like GTK+2.

now-im, 2017-06-26:
@lurch yes, only etcher always works and only in sudo etcher it fails. dev tools console output:
```
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 document not ready yet, trying again in 500 milliseconds...
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:53:59 GMT+0600 (+06) Application start
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:53:59 GMT+0600 (+06) Checking for updates ({"currentVersion":"1.0.0","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0","includeUnstableChannel":false})
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:54:07 GMT+0600 (+06) Update notification skipped ({"reason":"Latest version"})
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:54:36 GMT+0600 (+06) Open image selector
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:55:37 GMT+0600 (+06) Select image ({"path":"/run/media/nowshed/Softwares/Linux/manjaro-kde-minimal-17.0.2-pre6-testing-x86_64.iso","extension":"iso","size":{"original":1386045440,"final":{"estimation":false,"value":1386045440}},"logo":false,"bmap":false})
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:55:40 GMT+0600 (+06) Flash ({"image":"/run/media/nowshed/Softwares/Linux/manjaro-kde-minimal-17.0.2-pre6-testing-x86_64.iso","drive":{"device":"/dev/sdb","description":"Transcend 4GB","size":4048551936,"mountpoints":[{"path":"/run/media/nowshed/Drive1"}],"raw":"/dev/sdb","protected":false,"system":false,"name":"/dev/sdb"},"uuid":"9cb38ab1-abab-44a6-aa8d-5e15163a8b0f","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 WRITER: Re-spawning with elevation

/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Reading /proc/mounts
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: proc
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: sys
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: dev
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: run
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: efivarfs
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/sda8
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: securityfs
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: devpts
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: pstore
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: systemd-1
[... cut ...]
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:59:38 GMT+0600 (+06) Progress (write): 68% at 7.57 MB/s (eta 57s)
/tmp/.mount_flPD8w/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jun 26 2017 22:59:38 GMT+0600 (+06) Progress (write): 69% at 7.57 MB/s (eta 57s)
```
At this point it failed. Same thing with different percentage.

@jviotti from chromium 59, it depends on gtk3 exclusively now. Before it was a mix of gtk2 and gtk3. What i am seeing, electron in based on gtk3 now from the latest version.
https://www.archlinux.org/packages/community/x86_64/electron/

jviotti, 2017-06-30:
I've been digging into this a bit today. While Chromium does support GTK+3 by using a build flag, it doesn't seem be used in Electron at all.

- There is still an opened issue about moving to GTK+3 where the last comment is 15 days ago: https://github.com/electron/electron/issues/2927
- The Electron build instructions (https://electron.atom.io/docs/development/build-instructions-linux/#prerequisites) ask the users to install GTK+2 headers:

```
$ sudo apt-get install ... libgtk2.0-dev ...
$ sudo yum install ... gtk2-devel ...
$ sudo dnf install ... gtk2-devel ...
```

- Here is the C++ code of the dialogs, for reference: https://github.com/electron/electron/blob/master/atom/browser/ui/file_dialog_gtk.cc

I'm not sure why that AUR package lists gtk3 as a dependency. Could it be that that package is running on the cutting edge, enabling GTK+3 support even though its not mainstream?

Do you get the same results when running [the official AppImage we distribute in our website](https://resin-production-downloads.s3.amazonaws.com/etcher/1.0.0/Etcher-1.0.0-linux-x64.zip)?

jviotti, 2017-06-30:
OK, so the Etcher AUR package doesn't depend on that Electron AUR package that requires GTK+3. In fact, by looking at [`PKGBUILD`](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=etcher), the Etcher package depends on GTK+2:

```
depends=('gtk2' 'dbus' 'libxi' 'libxcursor' 'libxdamage' 'libxrandr'
	 'libxcomposite' 'libxext' 'libxfixes' 'libxrender' 'libxtst'
	 'gconf' 'glib2' 'nss' 'xz' 'fuse2')
```

jviotti, 2017-06-30:
I ran the Etcher latest AppImage on Fedora 24 with GNOME 3, and I don't see any difference in the open file dialogs:

## As a normal user

[image]

## With `sudo`

[image]

In fact, both dialogs seem to be GTK+2. Take a look a the open file dialog from Shotwell, a GTK+3 application:

[image]

jviotti, 2017-06-30:
I'll close since this looks like either:

1. Some weird GTK configuration on that KDE machine (I never used KDE, but I think they have "something" to make GTK apps look like QT ones, so maybe its not getting applied when running an with `sudo`)
2. Something weird with the AUR package

After quite some investigation, I'm pretty sure that the latest Etcher version is GTK+2 only.

Don't hesitate in re-opening if you find the reason and if there is something the Etcher project can do to fix it!

now-im, 2017-07-01:
As far as, I am seeing this may not be an etcher problem. By the way, gtk3 is totally mainstream right now, as gtk4 is already heavy development and maybe released at the end of the year. All major linux distro has made the shift.
Electron uses chromium webengine, so gtk3 will soon be the default. Arch package maintainer may have made the jump first. 
Good luck with the development, it has been a real nice alternative to SUSE image writer.

lurch, 2017-07-03:
@jviotti Ignoring (for now) all the discussion about GTK2 and GTK3, I still think it'd be interesting to try to dig into why for @now-im "only etcher always works and only in sudo etcher it fails." ?

@now-im Can you try re-running your Etcher writing tests with and without `sudo` using the AppImage version of Etcher, as @jviotti requested? Also, if you have an alternative USB drive / SD card adaptor you can _also_ test with, that might be interesting.

now-im, 2017-07-03:
@jviotti how do I run the official image?

jviotti, 2017-07-03:
@now-im You should be able to unzip, and execute the AppImage as a binary.

jviotti, 2017-07-03:
> @jviotti Ignoring (for now) all the discussion about GTK2 and GTK3, I still think it'd be interesting to try to dig into why for @now-im "only etcher always works and only in sudo etcher it fails." ?

Should we create a new issue for that? (or link it to an existing one)

now-im, 2017-07-03:
@jviotti I am getting this:
```
sudo sh Etcher-1.0.0-linux-x64.AppImage                                                                       
Etcher-1.0.0-linux-x64.AppImage: Etcher-1.0.0-linux-x64.AppImage: cannot execute binary file

```

lurch, 2017-07-03:
Why are you using `sh`? You want either `./Etcher-1.0.0-linux-x64.AppImage` or `sudo ./Etcher-1.0.0-linux-x64.AppImage`

Reopening issue (temporarily)

now-im, 2017-07-03:
My bad, i didn't know it doesn't have executing permission. This time the flash didn't get stuck but didn't work either because i can't mount my usb. Now, i will try to boot this as live image. This is the dev tool console output. By the way, this time also gtk2 based gdialog appeared.
```
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 document not ready yet, trying again in 500 milliseconds...
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:12:08 GMT+0600 (+06) Application start
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:12:08 GMT+0600 (+06) Checking for updates ({"currentVersion":"1.0.0","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0","includeUnstableChannel":false})
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:12:16 GMT+0600 (+06) Update notification skipped ({"reason":"Latest version"})
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:12:23 GMT+0600 (+06) Open image selector
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:13:22 GMT+0600 (+06) Select image ({"path":"/run/media/nowshed/Softwares/Linux/manjaro-my-kde-17.0.2-testing-x86_64.iso","extension":"iso","size":{"original":1410058240,"final":{"estimation":false,"value":1410058240}},"logo":false,"bmap":false})
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:13:24 GMT+0600 (+06) Flash ({"image":"/run/media/nowshed/Softwares/Linux/manjaro-my-kde-17.0.2-testing-x86_64.iso","drive":{"device":"/dev/sdb","description":"Transcend 16GB","size":15795748864,"mountpoints":[],"raw":"/dev/sdb","protected":false,"system":false,"name":"/dev/sdb"},"uuid":"5fa5a53a-1e6e-425c-939a-5ff8c7f2d537","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 WRITER: Re-spawning with elevation

/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Reading /proc/mounts
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: proc
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: sys
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: dev
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: run
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: efivarfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/sda8
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: securityfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: devpts
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: pstore
9/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: systemd-1
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: configfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: debugfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: mqueue
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: hugetlbfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/sda2
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
[... cut ...]
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:15:41 GMT+0600 (+06) Done ({"image":"/run/media/nowshed/Softwares/Linux/manjaro-my-kde-17.0.2-testing-x86_64.iso","drive":{"device":"/dev/sdb","description":"Transcend 16GB","size":15795748864,"mountpoints":[],"raw":"/dev/sdb","protected":false,"system":false,"name":"/dev/sdb"},"uuid":"5fa5a53a-1e6e-425c-939a-5ff8c7f2d537","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/tmp/.mount_BP6Hor/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Mon Jul 03 2017 22:15:41 GMT+0600 (+06) Navigate ({"to":"success","from":"main"})
```

lurch, 2017-07-03:
>  i can't mount my usb

That's probably because of something like https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

If the validation completed without reporting any errors, then that means that Etcher did its job correctly, so I guess the problem here must be something to do with the ArchLinux version, which was created by a 3rd party rather than us.  So I'll close this issue again, and I suggest you file a bug against the ArchLinux package of Etcher.

closed 2017-07-03 by lurch

now-im, 2017-07-03:
I have good news, usb booted, so you guys are clear. I think i will wait out a bit because electron will be gtk3 soon. This ought to fix the problem.