balena-io/etcher #1552: Display "Unmounting..." and "Elevating..." labels in the progress button

jviotti, 2017-06-27:
This would be very useful to quickly know at which point Etcher got stuck.

lurch, 2017-06-28:
I guess this ties in with #1438 too?

Shou, 2017-06-28:
Is the status when we're elevating exposed? Also we already have "Unmounting..." displayed, don't we?

jviotti, 2017-06-28:
Right, we only need to add "Elevating...." right before we try to spawn the elevation process. Regarding https://github.com/resin-io/etcher/pull/1438, let's get this merged first, and then send a new PR for the "Elevating..." addition.

Shou, 2017-06-29:
I'm guessing this mean we need to extend the flash state object?

lurch, 2017-06-29:
Sounds like `flash state` is slowly morphing into `etcher state`? ;-)

Shou, 2017-06-29:
Also, where does this leave "Starting..."?

jviotti, 2017-06-29:
> Sounds like flash state is slowly morphing into etcher state? ;-)

I guess that "elevating" is arguably a phase of a flash process.

> Also, where does this leave "Starting..."?

The flow would be the following:

- If the elevation dialog has to appear, show "Elevating..." right before spawning the elevation process
- Right before the writer process is spawned, "Starting..." (this will cover the case where diskpart cleans the drive on Windows for example, which takes a couple of seconds)
- Show "Unmounting..." right before starting to unmount the drive

What do you think?