balena-io/etcher #1556: cli without Electon

techtonik, 2017-06-28:
Etcher-cli-1.0.0-win32-x86.zip

It is surprising that this `dd` clone for Windows is 10Mb and doesn't run on older systems, because it requires GUI toolkit.

lurch, 2017-06-28:
It's already without a GUI toolkit and without Electron - but it does include a NodeJS interpreter.
And Etcher is far more than a 'dd clone'.

closed 2017-06-28 by lurch

techtonik, 2017-06-28:
> This application is only supported on Windows 7, Windows Server 2008 R2, or higher.

It if doesn't require Electron, why it doesn't work on other systems?

jhermsmeier, 2017-06-28:
@techtonik that's only supposed to clarify that Windows XP isn't supported anymore. It runs on a wide variety of Linux distributions, and Mac OS as well.

lurch, 2017-06-29:
I did a quick search, and found https://github.com/nodejs/node/issues/3804 and https://github.com/nodejs/node/pull/5167
If NodeJS itself doesn't support Windows XP, then the Etcher CLI (written in NodeJS) can't run on Windows XP either.

techtonik, 2017-06-29:
Yea. New NodeJS drops Windows. Too bad Etcher is not written in Go and depends on Windows 7 features that it doesn't use.

lurch, 2017-06-29:
Etcher is all Open Source, feel free to write your own clone of it in Go ;-)

Just out of curiosity, why do you need Etcher support on older versions of Windows? Are you able to try running it under Linux instead?

techtonik, 2017-07-02:
Because I used Windows in childhood and I am used to it and besides I still have 3 Linuxes running, I can not use them. I won't buy next Windows, but there is still no software with UX like Far Manager on Linux.

lurch, 2017-07-02:
> I won't buy next Windows

I'll wish you luck then, as more and more software is dropping support for Windows XP.

> there is still no software with UX like Far Manager on Linux.

That's something I'd never heard of, but a quick search found me: https://en.wikipedia.org/wiki/Far_Manager#Linux_and_MacOS_version

jhermsmeier, 2017-07-02:
@techtonik why not run Etcher on one of the Linuxes then?

techtonik, 2017-07-04:
@jhermsmeier good point. Still need to figure out the way to rebind Alt-F1 Alt-F2 on those Linuxes for my own purpose to port my console macros.