balena-io/etcher #1563: Both banners appear when forcing fullscreen

Shou, 2017-06-30:
When the window manager ignores window constraints and fullscreens Etcher, both the fallback and external banners show up. We could solve this by positioning the external banner from `top` instead of `bottom`.

[image]

lurch, 2017-06-30:
> When the window manager ignores window constraints

How common of a situation is that? ;)

jviotti, 2017-06-30:
@Shou My suggestion in the call was to wrap the contents of the application in a `div`, set a `max-height` and `max-width` to that `div`, and then center it in the window using flexbox. Then, if window manager maximizes the application, the content will remain centered, on the desired dimensions, so the only consequence will be a lot of margins.

Shou, 2017-07-03:
@lurch I believe most tiling WMs do this, so if you have a tiling WM then most of the time, unless you create rules for Etcher, e.g. make it float, or have a specific size.

@jviotti thanks for the refresher!

closed 2018-10-15 by jviotti