balena-io/etcher #1568: Debian repo not working!!!!

karl6994, 2017-07-03:
Please help!!! The " deb https://dl.bintray.com/resin-io/debian stable etcher " is not working!! I added the PGP key... and it just tells me its not working whenever I sudo apt update!!! :( 

E: The method driver /usr/lib/apt/methods/https could not be found.
N: Is the package apt-transport-https installed?
E: Failed to fetch https://dl.bintray.com/resin-io/debian/dists/stable/InRelease  
E: Some index files failed to download. They have been ignored, or old ones used instead.

- **Etcher version:none yet**
- **Operating system and architecture:Linux Debian 9 64-bit**
- **Image flashed:none yet**
- **Do you see any meaningful error information in the DevTools? Don't know what that is**

jhermsmeier, 2017-07-03:
Looks like you need to install `apt-transport-https` first, which you should be able to do by running
```
sudo apt-get update
sudo apt-get install apt-transport-https
```

karl6994, 2017-07-03:
Oh wow, never knew about that.. How strange, I've added these types of things before without that..
Thank you very much. It would be very helpful if this was included on the front github page by where they tell you to enter the repo..

lurch, 2017-07-03:
I see you're using the recently-released Debian 9 - perhaps that's changed some defaults around compared to Debian 8? I'll do some testing...

karl6994, 2017-07-26:
Not sure what's up with the current github repo & source-code release for 1.1.1 but its coming up as 1.0.0, app image comes up as 1.1.1 though. apologies I have no idea how to make one of these requests or I'd make a new one. Can't seem to find the button :S

lurch, 2017-07-26:
@karl6994 We've only just released Etcher v1.1.1, so you possibly caught things in the middle of the transition ;-)  Everything _should_ be pointing to 1.1.1 now, please let us know if you see any links which don't.

And on the topic of Debian 9, I did some testing just the other day and found I needed to install both `apt-transport-https` and `dirmngr` with `apt-get install apt-transport-https dirmngr` - I'll update the instructions soon (also planning to add instructions for Fedora at the same time).