balena-io/etcher #1572: [Suggestion] macOS Installer Support

KevinAT, 2017-07-06:
i don't think has been asked before. have you thought about adding the ability to burn macOS Installers with etcher?

jhermsmeier, 2017-07-06:
Hey @KevinAT, I'm not sure I understand this correctly; are you referring to the setup image that installs the operating system itself?

KevinAT, 2017-07-06:
@jhermsmeier yes, exactly those - like the windows os installer just the mac one. 

under OSX/macOS they are displayed as "Install OS X %cat_or_mountain_name%.app" and treated like normal executables. it's possible to burn them to dvds/usb drives/etc but from what i know in recent versions apple made it more difficult to burn them, as in requiring the app/image to be modified to be bootable. osx tools like Install Disk Creator and DiskMaker X 6 are able to create install mediums from the OSX Installer app files, but i think they modify them a bit in the background, and i'm not sure if those modifications would be easy to port to other OSs like linux. porting it to windows might not work at all, since ntfs doesn't seem to be able to handle the macOS installer files.

though i don't know much about this and am just speculating from what i heard/read. anyways it would be great if i could create bootable macOS installer drives in the future from my linux distro without having to use macOS myself.

jhermsmeier, 2017-07-06:
Okay – you can already do that, nothing complicated like preprocessing or modification necessary, as Etcher already supports the Apple Disk Image format.

### Step 1: Download Mac OS Whatever

<p align="center">
[image]
</p>

### Step 2: Navigate into `.app`'s package contents

<p align="center">
[image]
</p>

### Step 3: Find the `InstallESD.dmg` disk image

**NOTE:** It might be named differently for different Mac OS versions, but whatever has a `.dmg` extension in there and is ~4-6GB in size is what you're looking for.

<p align="center">
[image]
</p>

### Step 4: Flash it with Etcher

[image]

### Step 5: Done. That's it. That's all. 🎉

jhermsmeier, 2017-07-06:
> porting it to windows might not work at all, since ntfs doesn't seem to be able to handle the macOS installer files.

The host operating system doesn't matter, Mac OS' `.app`s are just glorified folders, which are treated differently by Mac OS, but you can just copy them to any other file system / storage device – so you should be able to do all this on any OS you like (apart from the downloading Mac OS from the App Store part, obviously).

jhermsmeier, 2017-07-06:
@KevinAT I'm going to close this for now, but do let us know if everything worked, or if not – what didn't!

closed 2017-07-06 by jhermsmeier

KevinAT, 2017-07-06:
@jhermsmeier really it's that simple? thanks a lot! i knew that there was that one big file that holds the osx image, but i didnt think etcher would be able to handle it if i just pulled it out of the app folder.

though i seem to have a few problems:
tried it on Win7-64 with both a sierra and a lion file:
etcher gets till the end of the process but then shows an error popup that always reappears when closed - had to force close the application.
[image]

and on macOS-Sierra with a sierra and a lion file:
doesn't even start the burning process and shows this popup straight away.
[image]

i could also try it on linux, but only when i get home.

it for sure isn't a problem with the files. i'm at work right now and copied the files from our file server, so they are definitely fine. and btw with the ntfs file issue before i meant that when trying to copy the whole macos installer.app file/folder over to an ntfs (windows) system it complains about the filename of CodeResources and OSInstallerCompatibility being wrong or too long. but since you said you only need the dmg file this shouldn't be a problem anyways.

also if etcher already knows where the ESD file would lie, why not make it possible to just select the "Install macOS xyz.app"-file/folder instead of making the user navigate all the way inside? i'm sure i can't be the only one who thought burning osx images wasn't possible because you couldn't select them with etcher. :)

lurch, 2017-07-11:
@KevinAT I believe the DMG support in Etcher was enhanced / fixed after the release of Etcher 1.0.0, so could you try the latest snapshot build?
https://resin-nightly-downloads.s3.amazonaws.com/etcher/2017-07-07/1.0.0%2Bfdceaad/Etcher-1.0.0%2Bfdceaad-darwin-x64.dmg

KevinAT, 2017-07-11:
@lurch tried it with that one - after asking for my user password it gives me an "Unmount failed - Error: Unmount failed"-error and after clicking ok, it gives me an Attention screen with the text "Oops, seems something went wrong." with the option to retry.

btw for the osx lion dmg it shows me something about it not being a bootable image and not having a partition table, is that supposed to be like that? doesn't complain about the sierra dmg not being bootable tho.

i think i'll reinstall my macos sierra system, to make sure the issue isn't on my end for some reason, and then try again.

lurch, 2017-07-11:
@jhermsmeier added all the DMG support, so I'll let him answer those questions...

jhermsmeier, 2017-07-11:
> i think i'll reinstall my macos sierra system, to make sure the issue isn't on my end for some reason, and then try again.

No need for such drastic measures – I'm seeing the same errors with our current nightly builds as well, and am currently debugging this. I'll let you know once I figure out what's happening here, and sorry for the mess!

> btw for the osx lion dmg it shows me something about it not being a bootable image and not having a partition table, is that supposed to be like that? doesn't complain about the sierra dmg not being bootable tho.

That warning can be safely ignored – Apple disk images sometimes have a padding at the start, and the partition table is further back, or they use an Apple Partition Map, which isn't read by Etcher (yet).

jhermsmeier, 2017-07-11:
I found the bug that was causing this, fixed it in the `udif` module, and updated it in https://github.com/resin-io/etcher/issues/1586.

I'm also running into "flash state percentage > 100" errors *again* with this, working on fixing that as well now.

KevinAT, 2017-07-12:
@jhermsmeier 

> No need for such drastic measures – I'm seeing the same errors with our current nightly builds as well, and am currently debugging this. I'll let you know once I figure out what's happening here, and sorry for the mess!

it's just a machine for testing stuff and when i occasionally need a mac, so reinstalling it wouldn't be a big issue. 
okay thx for your effort!

> That warning can be safely ignored – Apple disk images sometimes have a padding at the start, and the partition table is further back, or they use an Apple Partition Map, which isn't read by Etcher (yet).

good to know.