balena-io/etcher #1578: Error: spawn /bin/sh EAGAIN

firedev, 2017-07-10:
- **Etcher version:**
1.0.0
- **Operating system and architecture:**
macOS
- **Image flashed:**
I just left it open over a weekend, and noticed it now shows an error box
- **Do you see any meaningful error information in the DevTools?**
```
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:51
{"code":"EAGAIN","errno":"EAGAIN","syscall":"spawn /bin/sh","path":"sh","spawnargs":["-c","\"/var/folders/jz/2t5hs51j0rs17q6c3psdnlk00000gn/T/tmp-76428Eg5R0bc62QgQ.sh\""],"cmd":"\"/var/folders/jz/2t5hs51j0rs17q6c3psdnlk00000gn/T/tmp-76428Eg5R0bc62QgQ.sh\""}
```

[image]

lurch, 2017-07-10:
@jhermsmeier Do you think this might have the same cause as #1571 ?

jhermsmeier, 2017-07-10:
Could be the underlying cause, but EAGAIN in this case would hint at the spawn / child process timing out I believe – this is something we should handle gracefully, and as the error says, try again.

jviotti, 2017-07-10:
The `child_process.spawn()` function uses `uv_spawn()`, which is based on `fork()`, which says:

```
     [EAGAIN]           The system-imposed limit on the total number of processes under execution would be exceeded.  This limit is configura-
                        tion-dependent.

     [EAGAIN]           The system-imposed limit MAXUPRC (<sys/param.h>) on the total number of processes under execution by a single user
                        would be exceeded.
```

This hints me that there might be something keeping the temporary script processes up and running even when they are not longer needed (maybe the perl bug we fixed recently?), and therefore exhausting the system process limit during the weekend.

@firedev Did you notice high CPU usage after the weekend?

firedev, 2017-07-10:
@jviotti I can't say for sure, the notebook was mostly closed. I don't think I have noticed anything unusual today. Anything specific I should pay attention to the next time I see this?

jviotti, 2017-07-10:
@firedev I'd advice to open the Activity Monitor, and take a look at how many processes are active. Also take a look at the process list, to see if you can identify a single application with way too many processes (could be Etcher, or it could be something else.)

Sawaba, 2017-07-11:
@firedev - I'd also suggest looking at how many of these tmp-*.sh files are getting created in /var - that's how I noticed my issue. I left Etcher running overnight and had over 60,000 identical 6kb files the next day.

jviotti, 2017-07-12:
@Sawaba This PR (https://github.com/resin-io/etcher/pull/1574) should solve that particular issue. What OS are you in? We can send you the link to the snapshot build created by the CI services for that PR once it gets merged, so we can test if it made things better.

closed 2017-07-19 by jviotti