balena-io/etcher #1593: Opening .iso fails where file stored in blank named folder.

jchurc16, 2017-07-13:
1.0 - **Etcher version:**
Windows 7 x64 - **Operating system and architecture:**

I have an .iso file stored in a 'hidden' folder on my desktop, blank icon & renamed to ' ' (alt+0160) to keep the desktop clean. When trying to load the file, the below error is produced:

[image]

Opening & writing works fine when in any other folder, it looks like it's detecting this as an invalid file path.

jhermsmeier, 2017-07-13:
Looks like this is an instance of https://github.com/resin-io/etcher/issues/1459, and has already been fixed in #1559

jhermsmeier, 2017-07-13:
Also, as a side note; you can right-click on the desktop, then uncheck "Show desktop icons" under "View" in Windows if you want to avoid having to give your files & folders fancy invisible names.

jhermsmeier, 2017-07-13:
Closing as duplicate of #1459

closed 2017-07-13 by jhermsmeier