balena-io/etcher #1603: [minor bug]links are not working

lexoyo, 2017-07-17:
Hello

In the app, the links do not open a browser window. I click on them but nothing happens. I am talking about the links in the footer as well as the links to github and twitter after flashing.

Do you think it would be useful if I take the time and try to fix this myself?

- **Etcher version:** 1.0.0
- **Operating system and architecture:** linux fedora 23, 64bits
- **Image flashed:** ubuntu
- **Do you see any meaningful error information in the DevTools?** I do see this log which looks ok to me `Open external link ({"url":"https://github.com/resin-io/etcher"})`

lurch, 2017-07-17:
Any ideas what might be going wrong @Shou ?

jviotti, 2017-07-17:
Thanks for the report @lexoyo and thanks for offering your help! You can definitely give it a shot if you like. Feel free to ping me on Gitter if you need any guidance with the code/development process setup :)

Shou, 2017-07-17:
The Electron function `electron.shell.openExternal` [seems to use xdg-open](https://github.com/electron/electron/blob/db6d75ca90ebe23d70f40f767264f7c2d77f882e/atom/common/platform_util_linux.cc#L85) so make sure that `xdg-open 'http://google.com/'` in a terminal, for example, opens Google in a new tab in your browser. If it doesn't, you might have to use `update-alternatives` to configure it to do so.

jviotti, 2017-07-17:
Yeah, I just checked and seems to work fine for me too. Probably something related to @lexoyo 's configuration. I'll close, unless we can confirm its an issue with Etcher :)

closed 2017-07-17 by jviotti

lexoyo, 2017-07-17:
Ok, I do not see why but it does not make the software impossible to use so...

If there is someting easy I can do in a few hours, let me know