balena-io/etcher #1612: mini-bricked my usb stick

bebopalooblog, 2017-07-20:
- **Etcher version:** 1.0.0
- **Operating system and architecture:**  WIN10 x64 blah blah **
- **Image flashed:** AVLinux.iso
- **Do you see any meaningful error information in the DevTools?** N/A

I'd like to "de-etch" my usb stick, but I can't. All I can see in Win10 is the unformatted portion of the stick. I believe the first half has a CDFS. It boots to AVLinux in legacy-support mode, but GPARTED returns a recursive file system error while AVLinux is running. 

Runs live OK, but I'm trying to build a GPT install which isn't a current option for AVLinux and I'd like my memory back!

I don't think i needed Etcher in the first place - the iso has an efi directory - so I don't want to re-flash, but I'd like to reclaim the 4GB I've lost and I cant find a tool to do it. Windows disk utilities definitely won't handle it, and the Debian install I do have isn't mounting usb's at all at the moment.

Basically, Etcher seems to be the only utility which still recognizes the stick as 8GB (7.8). It would be nice to have a utility to reformat or wipe the whole stick. Is there somewhere you can plug one into your code?

Thanks,

Steve Moore
Canada

bebopalooblog, 2017-07-20:
Never mind - I just wiped the volume in win Disk Management and my memory returned :-)

Thanks!

closed 2017-07-20 by bebopalooblog

jviotti, 2017-07-20:
I'm glad you got it fixed. Windows disk management tools also give me tons of issues. The Etcher team usually uses `diskpart`, which (almost never :P) let us down, so I recommend to take a look at that if you want to perform these kinds of disk manipulation tasks.

lurch, 2017-07-20:
And see also https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives