balena-io/etcher #1613: stuck on 53% or 26% when writing to the flash

edinhui, 2017-07-20:
- **Etcher version:**
1.0.0
- **Operating system and architecture:**
Ubuntu 14.04
- **Image flashed:**
pcduino8_ubuntu14.04_20160102_preinstalled_opencv2.4.11.img
- **Do you see any meaningful error information in the DevTools?**

document not ready yet, trying again in 500 milliseconds...
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:19:24 GMT+0800 (CST) Application start
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:19:24 GMT+0800 (CST) Checking for updates ({"currentVersion":"1.0.0","releaseType":"PRODUCTION","updateSemverRange":"<2.0.0","includeUnstableChannel":false})
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:19:25 GMT+0800 (CST) Update notification skipped ({"reason":"Latest version"})
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:19:27 GMT+0800 (CST) Open image selector
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:19:40 GMT+0800 (CST) Select image ({"path":"/home/edinhui/Downloads/pcduino8_ubuntu14.04_20160102_preinstalled_opencv2.4.11.img","extension":"img","size":{"original":2592079872,"final":{"estimation":false,"value":2592079872}},"logo":false,"bmap":false})
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:19:45 GMT+0800 (CST) Flash ({"image":"/home/edinhui/Downloads/pcduino8_ubuntu14.04_20160102_preinstalled_opencv2.4.11.img","drive":{"device":"/dev/sdb","description":"Storage Device","size":31914983424,"mountpoints":[{"path":"/media/edinhui/3166-6430"}],"raw":"/dev/sdb","protected":false,"system":false,"name":"/dev/sdb"},"uuid":"746a3a3b-bfc8-4f8a-9e1a-5c6d3bbba3ca","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 WRITER: Re-spawning with elevation

/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Reading /proc/mounts
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: rootfs
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: sysfs
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: proc
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: udev
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: devpts
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: tmpfs
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/dm-1
8/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: none
9/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: cgroup
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/sda1
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: binfmt_misc
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: systemd
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: vmware-vmblock
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/dm-1
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: gvfsd-fuse
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: /dev/sdb1
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point belongs to drive
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Unmount success
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Mount point found: Etcher-1.0.0-linux-x64.AppImage
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 [mountutils] Closing /proc/mounts
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 0% at 11.99 MB/s (eta 215s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 0% at 12.23 MB/s (eta 210s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 1% at 13.34 MB/s (eta 192s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 1% at 13.21 MB/s (eta 194s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 1% at 13.78 MB/s (eta 185s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 1% at 13.63 MB/s (eta 187s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 2% at 14.01 MB/s (eta 181s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 2% at 13.86 MB/s (eta 182s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 2% at 13.47 MB/s (eta 187s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:07 GMT+0800 (CST) Progress (write): 2% at 13.4 MB/s (eta 188s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 3% at 13.66 MB/s (eta 183s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 3% at 13.59 MB/s (eta 184s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 3% at 13.51 MB/s (eta 185s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 3% at 13.19 MB/s (eta 189s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 4% at 12.77 MB/s (eta 195s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 4% at 12.75 MB/s (eta 194s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 4% at 12.53 MB/s (eta 197s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 4% at 12.52 MB/s (eta 197s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:08 GMT+0800 (CST) Progress (write): 5% at 12.53 MB/s (eta 196s)
[... cut ...]
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:09 GMT+0800 (CST) Progress (write): 53% at 10.54 MB/s (eta 115s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:09 GMT+0800 (CST) Progress (write): 53% at 10.55 MB/s (eta 114s)
/tmp/.mount_hPF8kG/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:51 Thu Jul 20 2017 20:26:09 GMT+0800 (CST) Progress (write): 53% at 10.56 MB/s (eta 114s)

now is 20:50:06, there is no update for about 24 minutes.

edinhui, 2017-07-20:
pcduino8_ubuntu14.04_20160102_preinstalled_opencv2.4.11.img is about 2.6GB, I don't know if it is related to the size.

jviotti, 2017-07-20:
Hi @edinhui ,

That's indeed very weird. I wonder what could have caused the write operations to hang without any error message.

Is this something you can reproduce every single time? If so, does it always hang at that particular percentage?

lurch, 2017-07-21:
I've just tested that image myself using both Etcher v1.0.0 and Etcher v1.1.0 and both were able to write it successfully. Is it possible that your SD card / Flash drive is faulty @edinhui ? Do you have a different card/drive you could try?

szotsaki, 2017-08-19:
I also have this problem on openSUSE Leap 42.3, x64. Mine was stuck at 15% for an hour or so. So I closed it. Console didn't show anything useful (apart from the regular status info [13% reached, 14% reached, 15% reached]).

SD card is fully functional, tested with [f3](http://oss.digirati.com.br/f3/).

lurch, 2017-08-19:
What image were you trying to write @szotsaki ?

> Edit: the interesting thing is, when 15% was reached, KDE shown that two "drives" (boot and the new OS itself) was mounted from the SD card.

Hmmm, sounds like the automounter is being a bit over-eager and mounting the partitions before we've finished writing the image :-/  Do you see the same behaviour if you try using `dd` instead of Etcher? (but if you've not used `dd` before, be very careful)

jviotti, 2017-08-21:
> Hmmm, sounds like the automounter is being a bit over-eager and mounting the partitions before we've finished writing the image :-/

@lurch We have an open descriptor to the block device, which should be preventing this. Are you aware of any other way to prevent mounting?

lurch, 2017-08-21:
@jviotti Hmmm, is there any way that the child-writer process could die, without Etcher detecting it? That _might_ explain why the devtools console (and Etcher's progress bar) stops updating, and why the open-descriptor to the block device gets released (allowing the automounter to then kick in) ?

jviotti, 2017-08-21:
I double checked that we're handling child process exits correctly (and we do), but I now have another theory: this could happen if the IPC channel stops for some reason.

jviotti, 2017-08-21:
And I wonder if the culprit is https://github.com/resin-io/etcher/blob/master/lib/child-writer/writer-proxy.js#L121

utrescu, 2017-09-18:
Same problem on Fedora 26. Stuck at a random % ( 75%, 80%, ... ) 

There are no relevant info in logs.

jhermsmeier, 2017-11-17:
This should be fixed by https://github.com/resin-io/etcher/pull/1843

closed 2017-11-17 by jhermsmeier

PilnyTomas, 2022-02-28:
By any chance, could this be power-related? I had the same issue (always stopped around 42%).
At first, I had the USB card reader connected via USB hub in which was already a number of other devices connected.
I tried to connect the USB card reader directly to the port without any other device drawing from that port and it worked the first time.