balena-io/etcher #1614: Full size 8gb img won't burn using Etcher on Windows

mkeegan, 2017-07-20:
- **Etcher version:** 1.0.0
- **Operating system and architecture:** Windows 10 Home
- **Image flashed:** Kano Pixel Kit Firmware
- **Do you see any meaningful error information in the DevTools?** No

Using a 7.9GB (7580 * 1024 * 1024 bytes) .img file with a Sandisk Ultra 8GB Class 10 Micro SD card.

Etcher confirms card as 7.9GB. Burning proceeds correctly until 99% where it stays for approximately 4 minutes. An error message then appears saying there is insufficient space available on the SD card.

A check of Windows disk management reports card size as 7.4 GB.

Performed burn sequence using same SD card and .img on MacOS 10.12.3 with Etcher 1.0.0, which completed successfully.

Tested same SD card on Windows with Win32DiskImager successfully.

I suspect that the problem is due to the fact that Windows calculates a megabyte as 1024 * 1024 bytes, where as MacOS declares it as 1000 * 1000 bytes. This would be why 7.4 * 1024 * 1024 * 1024 = 7.9 GB of actual storage space.

lurch, 2017-07-20:
Good news! This should already have been fixed by #1574 (which also contains links to test builds, if you'd like to confirm for us?)

jviotti, 2017-07-21:
Just to provide some more context: we used to get the drive size through WMI, which was known to not provide a very accurate number, which brought all sorts of other issues. 1.1.0 does the same thing using a Windows IOCTL that provides the real exact number, so I think it should work there.

lurch, 2017-07-21:
...and since I added my comment yesterday, Etcher v1.1.0 was released which contains this fix too :)

lurch, 2017-07-21:
@jviotti Hmmm, if this was due to the WMI-size bug we fixed (in drivelist), wouldn't Etcher (on Windows) think that the card is smaller than it _actually_ is, and therefore refuse to attempt to write the image in the first place?
But re-reading the original message, he says that Etcher _was_ able to start burning the image, but then failed at 99%...

mkeegan, 2017-07-21:
@lurch you're correct in that Etcher did work for the first burn attempt. All subsequent burn attempts displayed a "Not enough Space" message until the card was re-formatted in another machine.

lurch, 2017-07-21:
Huh? That's really weird - even WMI should always report the same card size, regardless of whatever data is or isn't on the card :confused: There should never be any need to re-format a card to get Etcher to see it's "true" size.

Is the image you used available for download anywhere? Is it possible that you have multiple SD cards with slightly different sizes (one just bigger than the image, and one just smaller) and you're just getting them muddled up? ;-)
Are you able to re-test with Etcher v1.1.0 to see if it has the same behaviour?

mkeegan, 2017-07-21:
@lurch unfortunately I can't give the image out at the moment 😞 

Definitely using the same card since the testing team tag them as they enter the lab.

I'll re-test with the new version next week.

mkeegan, 2017-07-27:
@lurch test builds were unsuccessful. They both crashed soon into the burning process. I will grab some screenshots for you today.

lurch, 2017-07-27:
In the meantime, Etcher v1.1.1 has been released, so please test with that, and send screenshots as appropriate ;-)

jhermsmeier, 2017-07-27:
@mkeegan In the screenshot you're flashing a `.img.gz` – what's the size of that, and what's the size of the raw, uncompressed image? I'm asking because gzip stores the uncompressed size as `size % 2^32`, and thus isn't able to encode sizes of 4GB and above correctly.

mkeegan, 2017-07-27:
@jhermsmeier the screenshot was actually from a customer, so I'm not too sure on that one, sorry.

The size of our .img is being reported by Windows 10 as 7.4GB (7,948,206,080 bytes). We attempted with the both compressed and uncompressed files.

lurch, 2017-07-27:
@mkeegan It would be interesting if you could also run https://github.com/resin-io-modules/drivelist on both Windows and MacOS, to see what each of them thinks the raw SD card size is.

jhermsmeier, 2018-01-16:
Closing, as this issue should be resolved in later versions – if not, feel free to re-open

closed 2018-01-16 by jhermsmeier