balena-io/etcher #1615: Etcher-cli: list of flashable devices

tangb, 2017-07-20:
Hi

I want to use etcher-cli on linux mac and windows and I'm looking for a way to detect flashable devices on all platforms.
Is etcher-cli able to return available flashable devices?

Thank you for your answer ;)

lurch, 2017-07-21:
Not at the moment, but https://github.com/resin-io-modules/drivelist is able to do that :)

lurch, 2017-07-24:
See also https://github.com/resin-io/etcher/issues/1420#issuecomment-301581820

lurch, 2018-02-26:
@tangb Is `drivelist` able to provide you the info you're looking for (in which case can this issue be closed?), or is there something more specific you're looking for from etcher-cli?

jhermsmeier, 2018-02-26:
@lurch I think the intention is to list the available drives from the CLI, to see which one to flash. But that's also already available; if you omit the --drive option you'll be presented with a list of devices to select from, i.e.:

```
$ etcher ~/Downloads/something.img
? Select drive (Use arrow keys)
❯ /dev/disk2 (31.9 GB) - SD Card Reader
```

Closing, as this feature is already available

closed 2018-02-26 by jhermsmeier

tangb, 2018-02-27:
@jhermsmeier  My first intention was to list available drives from python. So i was looking for if etcher-cli was able to return me that list.
Finally I find another way to do the job, looking how it was done in drivelist.
Anyway the "feature" can be closed

Thank you

jhermsmeier, 2018-02-27:
Sorry that I misunderstood; in that case you're really looking for [drivelist](https://github.com/resin-io-modules/drivelist) 😄