balena-io/etcher #1617: Etcher 1.1.0, Flash do not start the flashing process

AlexisTM, 2017-07-21:
- **Etcher version:** 1.1.0
- **Operating system and architecture:** Linux Ubuntu 14.04 x64
- **Image flashed:** .img.xz

When I try to flash the image with Etcher 1.1.0, when I press "flash" and I put the password, the button says Starting... then become the blue "Flash" again. 

Flashing the same image on the same SDCard works perfectly on Etcher 1.0.0.

lurch, 2017-07-21:
I tried reproducing this myself, with an `.xz` compressed image, and everything worked fine (using the `.deb` installed version of Etcher v1.1.0).

Then I tried using the AppImage version of Etcher v1.1.0, and I see the same behaviour as you do (even with an uncompressed image)  :-(
When I look in the DevTools console I see a message saying `Elevation cancelled` even though I entered the correct password.

I then tried running the AppImage with `sudo`, but this time it just 'locks up' at the `Starting...` stage, and the DevTools console is stuck at `WRITER: Re-spawning with elevation`.
.......oh! And then after being stuck for ages, it finally finishes writing the disk almost instantly and the DevTools console suddenly gets all the logs at once (so it looks like the write _is_ happening in the background, but Etcher doesn't 'catch up' until after the process has finished).

Until we fix this, it looks like the only workaround is to use the `.deb` version of Etcher v1.1.0 (if you're using a Debian or Ubuntu based OS), or revert back to using Etcher v1.0.0.

AlexisTM, 2017-07-21:
Indeed, it was the AppImage version. I forgot to mention.

lurch, 2017-07-21:
...and I've just tried the same with a fresh Linux x64 AppImage built from the head of `master` (with `make electron-develop && make electron-installer-appimage`), and that behaves the same as the Etcher v1.1.0 AppImage :-/  But this definitely seems to be an AppImage-specific bug because if (in my build directory) I do `cd dist/linux-unpacked/; ./etcher-electron` then that _is_ able to flash an image successfully.

Any ideas if this might be a bug in `electron-builder` @develar ?

develar, 2017-07-21:
@lurch Without error message it is not easy for me to advice. electron-builder appImage is quite simple and I don't see place for such error. If you mean that error appeared after migration to electron-builder — maybe problem that now asar is used (was it used before electron-builder?). In this case try to disable asar to check.

lurch, 2017-07-21:
I'll try bisecting to work out when the problem started...

jviotti, 2017-07-24:
I confirm this is fixed by https://github.com/resin-io/etcher/pull/1625. @develar The problem seems to be the AppImages version `electron-builder` is using. AppImages v6 seems work fine. Is it easy to upgrade? Can I help somehow with it?

jviotti, 2017-07-24:
@AlexisTM , @ddulic, @CodeGradox Sorry for the inconveniences. Please use the deb or rpm packages for now. We'll be re-publishing the AppImage very soon.

lurch, 2017-07-27:
Etcher v1.1.1 fixes this problem.

closed 2017-07-27 by lurch